Ultimaker/Cura #8791: slow down support contact layers

cruzaderz, 2020-11-18:
Hi Team,

as I repetedly fail to print the corona virus in one piece and the small contact points seem the reason my wish:

It should be possible to set "slow down x under layers" and "...x over layers" so after the support the printer may build a nice post to go on. In my case the support interface itself is set as slow by option but right over the printer goes on with usual 200 mm/s. And that causes that 1*1 mm blob it starts to stick too less so it can't support stuff coming on top.

Regards
Dennis

Ghostkeeper, 2020-11-19:
No, the best you can currently do is to reduce the Top/Bottom Speed. Isn't 200mm/s a bit excessive? You need to have a pretty long heat zone to achieve that, I'd think!

There is also a setting called Overhanging Wall Speed, but it explicitly doesn't affect walls that get supported by support.

We already have a feature on our backlog to add a bottommost-skin layer with different settings, which would include this feature request. So this is planned to be implemented then. It doesn't have great priority, but it's on the agenda at least.

cruzaderz, 2020-11-22:
Ah Top/Bottom is 100 mm/s already because exceeding it the skin is not flat enough for my taste.
The 200 mm/s affect mainly infill and inner walls; outer walls are done at 70 mm/s.
I dont know much about the heat zone but my Titan Aero is described having a very short one.
Guess more it affects that I print 0.3 mm nozzle at 0.4 width and 0.08 layers.
So compare that heat question to a <100 mm standard line (0.48/0.2) and yep, works fine... 👍 

But coming back to the initial issue:
A speed reduction at the contact layers would be great as often the first dot of the supported layer is just a 1 mm "pimple" that doesnt really support what comes on top.
A refuction in speef could make it stick better and make the shape better so it would not be kicked off so easy.

Regards
Dennis