Ultimaker/Cura #8792: Bad slicing objetc coming from Fusion360

Chrispizz, 2020-11-18:
**Application version**
CURA 4.8.0

**Platform**
Windows 10 / Radeon R9 380 

**Printer**
ANET A6

**Reproduction steps**
Just slice the STL objet and slice it , then go to PREVIEW
[CUBE Fusion360.zip](https://github.com/Ultimaker/Cura/files/5559899/CUBE.Fusion360.zip)

**Screenshot(s)**
See attached screenshot 
[image]

**Actual results**
After slice step , the holes is filled by threads as it adds not existing thing inside.

**Expected results**
The holes should be empty , without matter inside ( see the difference between PREVIEW and PREPARE mode )

**Project file**
You will find the project in attachment too 
[CURA Project CUBE Fusion360.zip](https://github.com/Ultimaker/Cura/files/5559931/CURA.Project.CUBE.Fusion360.zip)

**Log file**
Log file is here 
[cura.log](https://github.com/Ultimaker/Cura/files/5559950/cura.log)

**Additional information**
No more to say about this , I hope you will find the solution ...

Many thank's 

Chrispizz

nallath, 2020-11-18:
You have "Make Overhang Printable" enabled.

closed 2020-11-18 by nallath

Chrispizz, 2020-11-18:
Nallath , Many thanks for this 
I'm going to coorect and try