Ultimaker/Cura #8793: pls help Cura crashing after loading gui have done a fresh install still crashes

TheLastGriffin8, 2020-11-18:
2020-11-18 23:27:41,955 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-11-18 23:27:41,987 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-11-18 23:27:42,007 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2020-11-18 23:27:42,024 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2020-11-18 23:27:42,043 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2020-11-18 23:27:42,064 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2020-11-18 23:27:42,106 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
2020-11-18 23:27:42,131 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeWriter
2020-11-18 23:27:42,162 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImageReader
2020-11-18 23:27:42,186 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-11-18 23:27:42,209 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalContainerProvider
2020-11-18 23:27:42,237 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalFileOutputDevice
2020-11-18 23:27:42,258 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MachineSettingsAction
2020-11-18 23:27:42,274 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ModelChecker
2020-11-18 23:27:42,293 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MonitorStage
2020-11-18 23:27:42,320 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PerObjectSettingsTool
2020-11-18 23:27:42,343 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PostProcessingPlugin
2020-11-18 23:27:42,363 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2020-11-18 23:27:42,379 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PreviewStage
2020-11-18 23:27:42,409 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RemovableDriveOutputDevice
2020-11-18 23:27:42,425 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2020-11-18 23:27:42,469 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimulationView
2020-11-18 23:27:42,491 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SliceInfoPlugin
2020-11-18 23:27:42,515 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2020-11-18 23:27:42,537 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2020-11-18 23:27:42,630 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-11-18 23:27:42,633 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-11-18 23:27:42,635 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-11-18 23:27:42,636 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-11-18 23:27:42,638 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-11-18 23:27:42,640 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-11-18 23:27:42,641 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-11-18 23:27:42,643 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-11-18 23:27:42,650 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin Toolbox
2020-11-18 23:27:42,673 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CameraTool
2020-11-18 23:27:42,698 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MirrorTool
2020-11-18 23:27:42,725 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RotateTool
2020-11-18 23:27:42,753 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ScaleTool
2020-11-18 23:27:42,772 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SelectionTool
2020-11-18 23:27:42,798 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TranslateTool
2020-11-18 23:27:42,817 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TrimeshReader
2020-11-18 23:27:42,838 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPReader
2020-11-18 23:27:42,859 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPWriter
2020-11-18 23:27:42,883 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UltimakerMachineActions
2020-11-18 23:27:43,189 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UM3NetworkPrinting
2020-11-18 23:27:43,215 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [40]: Checking for new version of cura
2020-11-18 23:27:43,215 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UpdateChecker
2020-11-18 23:27:43,510 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin USBPrinting
2020-11-18 23:27:43,675 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade21to22
2020-11-18 23:27:43,739 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade22to24
2020-11-18 23:27:43,808 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade25to26
2020-11-18 23:27:43,882 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade26to27
2020-11-18 23:27:43,943 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade27to30
2020-11-18 23:27:44,006 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade30to31
2020-11-18 23:27:44,079 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade32to33
2020-11-18 23:27:44,126 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade33to34
2020-11-18 23:27:44,194 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade34to35
2020-11-18 23:27:44,226 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade35to40
2020-11-18 23:27:44,248 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade40to41
2020-11-18 23:27:44,270 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade41to42
2020-11-18 23:27:44,292 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade42to43
2020-11-18 23:27:44,310 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade43to44
2020-11-18 23:27:44,331 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade44to45
2020-11-18 23:27:44,350 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade45to46
2020-11-18 23:27:44,370 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade460to462
2020-11-18 23:27:44,390 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade462to47
2020-11-18 23:27:44,410 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade47to48
2020-11-18 23:27:44,428 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimpleView
2020-11-18 23:27:44,453 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin X3DReader
2020-11-18 23:27:44,489 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin XmlMaterialProfile
[... 44 lines cut ...]
2020-11-18 23:27:59,516 - DEBUG - [Thread-2] cura.Machines.ContainerTree.run [174]: Started background loading of MachineNodes
2020-11-18 23:27:59,518 - DEBUG - [Thread-2] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed
2020-11-18 23:28:00,600 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2020-11-18 23:28:00,602 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [74]: No active GlobalStack, set quality profile model as empty.
2020-11-18 23:28:00,604 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2020-11-18 23:28:00,606 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [38]: No active GlobalStack, set CustomQualityProfilesDropDownMenuModel as empty.
2020-11-18 23:28:00,844 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [210]: Support for Vertex Array Objects: True
2020-11-18 23:28:00,864 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2020-11-18 23:28:00,866 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1.0 NVIDIA 457.30
2020-11-18 23:28:00,868 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   NVIDIA Corporation
[... cut ...]
2020-11-18 23:28:00,944 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray_composite.shader]...
2020-11-18 23:28:00,953 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\platform.shader]...
2020-11-18 23:28:00,965 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray.shader]...

Ghostkeeper, 2020-11-19:
Duplicate of https://github.com/Ultimaker/Cura/issues/8029. That same weird message in the log too, about the preferences being corrupted or not found, indicating that maybe it can't write the preferences file. Might have something to do with access rights to the configuration folder?

closed 2020-11-19 by Ghostkeeper

TheLastGriffin8, 2020-11-20:
cura will only open if i uninstall Nvidia Game ready Driver v457.30 but it starts crashing if i re install it and i kinda need it. if it helps at all

TheLastGriffin8, 2020-11-20:
i found a temp fix i had to downgrade nvidia gme ready driver from 457.30 to 456.38 guess something int he newer driver broke it.