Ultimaker/Cura #8797: Seperate Jerk settings for X and Y axis

ComedianTF2, 2020-11-19:
**Is your feature request related to a problem? Please describe.**
Some printer models use separate jerk settings for the X and the Y axis. In Cura, you can not set these separately for the X and Y axis, even though it is possible to set these differently in using gcode settings. To achieve accurate printing estimation times in cura, you need to enable jerk control to get these settings accurately.

While it's acceptable that the difference in jerk settings between the two axis is slightly different for the time calculation, for printing results it means either a slightly slower print by setting it to the lower of the two settings, or slightly worse printing results by setting it too high.

**Describe the solution you'd like**
Separate X and Y values

**Describe alternatives you've considered**
The current way is the alternative. It could be hidden behind an option, to not confuse people

**Affected users and/or printers**
Only printers that have alternate jerk settings for the X and Y axis are affected. This can be printers that have a large and heavy moving bed on the Y axis and only the hotend on X or the like

Ghostkeeper, 2020-11-23:
Hmm, I don't see that in the documentation here: https://reprap.org/wiki/G-code#M205:_Advanced_settings Not yet, at least. Maybe this is a new feature that hasn't been documented yet, or maybe it's not in a significant amount of printers yet as to warrant their own category.

You'd expect to double those jerk settings to have a separate setting for Jerk in X for infill and Jerk in Y for infill? And wouldn't that then also hold for acceleration and speed?

Currently the jerk settings are some of the least-adjusted settings in the whole list, so splitting those up might also just be busywork that nobody is going to use. The difficulty in this is to get the print time estimation right.

ComedianTF2, 2020-11-23:
It's possible under the M566 settings, rather than under M205: https://reprap.org/wiki/G-code#M566:_Set_allowable_instantaneous_speed_change

Cura uses M566 when the gcode flavour is set to RepRapFirmware, and M205 when set to marlin.

See also https://duet3d.dozuki.com/Wiki/Gcode#Section_M566_Set_allowable_instantaneous_speed_change

It should indeed also be possible for acceleration and speed

Ghostkeeper, 2020-11-26:
This could be built rather invisibly to non-RepRap users by hiding the X/Y specialised settings if the g-code flavour is not RepRap.

However since Ultimaker printers nor Marlin support this, the implementation of this is rather difficult. Our time estimation algorithm is based on Marlin's behaviour, which is similar to most printers in the motion aspect. If this implementation is only in a different firmware that's very hard to align with our current time estimation algorithm. Furthermore, since Ultimaker's printers don't implement this there is not going to be room in our planning to implement this; I can't justify spending a significant amount of time in adjusting the time estimation algorithm if it's not for an Ultimaker printer.

I think we'll accept contributions that adjust this if they do not break the existing behaviour for the time estimation. But don't expect an implementation coming from Ultimaker to implement this.

As a workaround, you can adjust this in the start g-code (and then disable jerk control by Cura itself) or through a search-and-replace script that adds something after the `;TYPE:INFILL` comments and such in the output g-code.

ComedianTF2, 2020-11-26:
I think that's a fair resolution, thanks for looking into it

GregValiant, 2023-07-02:
See pull request #15954