Ultimaker/Cura #8801: [4.8.0] Gapfill placed between wall lines

psycowithespn, 2020-11-20:
**Application version**
4.8.0 (issue NOT in 4.7.1)

**Platform**
Windows 10 2004, Nvidia 980ti

**Printer**
Creality Ender 3

**Reproduction steps**
1. Have "fill gaps between walls" set to everywhere.
2. Slice a model with a fairly specific curve radius (depending on the maximum resolution).

**Screenshot(s)**
4.8.0 (yellow lines around corners of right cube and between walls of cylinders)
[image]

4.7.1 (no yellow lines between walls)
[image]

**Actual results**
Gapfill is placed in between wall lines and the print time is extended slightly. I haven't printed with this yet, but I would expect to see some extra zits.

**Expected results**
Gapfill should not be placed between wall lines.

**Project file**
[Cura Issue.zip](https://github.com/Ultimaker/Cura/files/5570704/Cura.Issue.zip)

**Log file**
Not a crash.

**Additional information**
I initially noticed this with the ripple test model (from [here](https://www.thingiverse.com/thing:277394)). I was able to create a more minimal SolveSpace project that can reproduce this (with "export chord tolerance (in mm)" set to 0.02). It was a bit finicky to get the radius just right.

The only settings I see changed between the 2 versions were "Maximum Resolution" and "Maximum Travel Resolution". They were 0.05 in 4.7.1 and 0.25 in 4.8.0. However, setting 4.7.1 to 0.25 does NOT cause the issue. Setting 4.8.0 to 0.05 fixes the issue, I assume by moving the issue to a tighter curve radius.

Infill options don't seem to affect it. Setting it to 0% infill doesn't resolve the issue.

Setting "fill gaps between walls" to "nowhere" gets rid of the extra extrusion, which is why I think this is gapfill.

Ghostkeeper, 2020-11-23:
Of course, the gap filling is intended to fill gaps between walls in sharp corners.
[image]

But indeed for this print that is unnecessary. You can remove them (while keeping gap filling enabled) by enabling "Filter Out Tiny Gaps".

The problem is that indeed there are actual gaps between the wall there due to the reduction in resolution for the inner walls. It's theoretically correct, but in practice you wouldn't want it to go back to those places and fill up those gaps. Maybe it needs to filter out some gaps even if the Filter Out Tiny Gaps is disabled, for a less correct result but better print quality.

Ghostkeeper, 2020-11-24:
This will be fixed by Ultimaker/CuraEngine#1210 , our upcoming implementation of variable line width. In that branch we're not filling gaps between walls any more with separate lines. We're filling it as part of the wall generation with singular central wall lines.

With the workaround being simple (enable Filter Out Tiny Gaps), and Filter Out Tiny Gaps being enabled by default for most printers, we've decided that we're not going to spend time on a fix for the intermediate version (as 4.9 will not have that variable line width feature yet).

Fredrossi046, 2021-01-22:
The option of "enable Filter..." and "Filter out..." does not give me the desired fix in 4.8.0. My problem in the end is that the printer starts skipping because of those yellow lines. I would have to lower my line-width to get rid of them. Unfortunately this will sometimes create gaps. See attached pictures.

normal line-width (0.4 as nozzle)
[image]

Line-width of 0.37
[image]

Ghostkeeper, 2021-02-01:
Lowering the line width is (in my opinion) the best solution for such cases. I've explained the targets to go for here, although I think that's something you'll already know: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/resolution/line_width.md#adjusting-line-widths-to-fit-enough-walls

It's not removing that gap filling because it's not tiny enough. The threshold is 2 square line widths there, and this is not configurable. That's a limitation of Cura right now.

One thing you are missing, I think, is that there is theoretically no gap in the second picture (aside from the little lip at the bottom of the picture). What happens is that Cura draws 6 contours for this ring, of which the middle 2 overlap a lot. Due to the setting Compensate Inner Wall Overlaps, it's going to calculate how much these two innermost walls overlap, and reduce the flow rate of one of them to match the remaining non-overlapping volume precisely. The position of that line is not adjusted though, so it gets printed 0.15mm or so towards the side, but this material is meant to flow into the gap precisely, leaving no gap.
Compensate Wall Overlaps is also replaced by the variable line width implementation in Ultimaker/CuraEngine#1210.

Ghostkeeper, 2021-12-14:
This is fixed by https://github.com/Ultimaker/CuraEngine/pull/1210.

closed 2021-12-14 by Ghostkeeper