Ultimaker/Cura #8803: installing pynest2d

EminFaruk, 2020-11-20:
I am trying to install Cura on Windows 10 using VS 2015. I am using the instructions from https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows

I was able to follow all the steps with no problem until this point. At installing Pynest2D at step 10, when I run the nmake command. I get the following error:

```
[100%] Linking CXX shared library pynest2d.pyd
LINK: command "E:\Program Files\Microsoft Visual Studio 14.0\VC\bin\amd64\link.exe /nologo @CMakeFiles\python_module_pynest2d.dir\objects1.rsp /out:pynest2d.pyd /implib:_pynest2d.lib /pdb:E:\DEV\pynest2d\build\pynest2d.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO C:\Users\HP\AppData\Local\Programs\Python\Python36\libs\python36.lib E:\DEV\libnest2d\build\dependencies\lib\polyclipping.lib E:\DEV\libnest2d\build\dependencies\lib\nlopt.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:pynest2d.pyd.manifest" failed (exit code 1120) with the following output:
   Creating library _pynest2d.lib and object _pynest2d.exp
```

Can anyone suggest how to solve this problem?

Thank you.

Ghostkeeper, 2020-11-24:
Hmm, so the error message indicated that `link.exe` failed. But gives no error message as to how or why it failed to link. No idea.

The pynest2d library is rather new so those instructions have not been refined/hardened to work on many various systems yet. By nature we can only write them down as to what works in theory and what works on our computers. I hope that through bug reports like this we can create instructions that work on more computers.

EminFaruk, 2020-11-24:
Dear Ghostkeeper,

You can see the whole error code I get below.

I get this error, after successfully installing the libnest2d, while trying to install pynest2d. 

I would be very glad if anyone can help, since I am new to this code. Thanks.

```
LINK: command "E:\Program Files\Microsoft Visual Studio 14.0\VC\bin\amd64\link.exe /nologo @CMakeFiles\python_module_pynest2d.dir\objects1.rsp /out:pynest2d.pyd /implib:_pynest2d.lib /pdb:E:\DEV\pynest2d\build\pynest2d.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO C:\Users\HP\AppData\Local\Programs\Python\Python36\libs\python36.lib E:\DEV\libnest2d\build\dependencies\lib\polyclipping.lib E:\DEV\libnest2d\build\dependencies\lib\nlopt.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:pynest2d.pyd.manifest" failed (exit code 1120) with the following output:
   Creating library _pynest2d.lib and object _pynest2d.exp
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_create referenced in function "public: __cdecl nlopt::opt::opt(enum nlopt::algorithm,unsigned int)" (??0opt@nlopt@@QEAA@W4algorithm@1@I@Z)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_destroy referenced in function "protected: struct libnest2d::opt::Result<double> __cdecl libnest2d::opt::NloptOptimizer::optimize<class <lambda_2553785388a20e04dc430272b2ba8569> &,double>(class <lambda_2553785388a20e04dc430272b2ba8569> &,class std::tuple<double>,class libnest2d::opt::Bound<double>)" (??$optimize@AEAV<lambda_2553785388a20e04dc430272b2ba8569>@@N@NloptOptimizer@opt@libnest2d@@IEAA?AU?$Result@N@12@AEAV<lambda_2553785388a20e04dc430272b2ba8569>@@V?$tuple@N@std@@V?$Bound@N@12@@Z)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_copy referenced in function "protected: struct libnest2d::opt::Result<double> __cdecl libnest2d::opt::NloptOptimizer::optimize<class <lambda_2553785388a20e04dc430272b2ba8569> &,double>(class <lambda_2553785388a20e04dc430272b2ba8569> &,class std::tuple<double>,class libnest2d::opt::Bound<double>)" (??$optimize@AEAV<lambda_2553785388a20e04dc430272b2ba8569>@@N@NloptOptimizer@opt@libnest2d@@IEAA?AU?$Result@N@12@AEAV<lambda_2553785388a20e04dc430272b2ba8569>@@V?$tuple@N@std@@V?$Bound@N@12@@Z)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_optimize referenced in function "public: enum nlopt::result __cdecl nlopt::opt::optimize(class std::vector<double,class std::allocator<double> > &,double &)" (?optimize@opt@nlopt@@QEAA?AW4result@2@AEAV?$vector@NV?$allocator@N@std@@@std@@AEAN@Z)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_set_min_objective referenced in function "protected: struct libnest2d::opt::Result<double> __cdecl libnest2d::opt::NloptOptimizer::optimize<class <lambda_2553785388a20e04dc430272b2ba8569> &,double>(class <lambda_2553785388a20e04dc430272b2ba8569> &,class std::tuple<double>,class libnest2d::opt::Bound<double>)" (??$optimize@AEAV<lambda_2553785388a20e04dc430272b2ba8569>@@N@NloptOptimizer@opt@libnest2d@@IEAA?AU?$Result@N@12@AEAV<lambda_2553785388a20e04dc430272b2ba8569>@@V?$tuple@N@std@@V?$Bound@N@12@@Z)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_set_max_objective referenced in function "protected: struct libnest2d::opt::Result<double> __cdecl libnest2d::opt::NloptOptimizer::optimize<class <lambda_2553785388a20e04dc430272b2ba8569> &,double>(class <lambda_2553785388a20e04dc430272b2ba8569> &,class std::tuple<double>,class libnest2d::opt::Bound<double>)" (??$optimize@AEAV<lambda_2553785388a20e04dc430272b2ba8569>@@N@NloptOptimizer@opt@libnest2d@@IEAA?AU?$Result@N@12@AEAV<lambda_2553785388a20e04dc430272b2ba8569>@@V?$tuple@N@std@@V?$Bound@N@12@@Z)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_get_algorithm referenced in function "public: enum nlopt::algorithm __cdecl nlopt::opt::get_algorithm(void)const " (?get_algorithm@opt@nlopt@@QEBA?AW4algorithm@2@XZ)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_get_dimension referenced in function "private: void __cdecl nlopt::opt::alloc_tmp(void)" (?alloc_tmp@opt@nlopt@@AEAAXXZ)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_get_errmsg referenced in function "public: char const * __cdecl nlopt::opt::get_errmsg(void)const " (?get_errmsg@opt@nlopt@@QEBAPEBDXZ)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_set_lower_bounds referenced in function "public: void __cdecl nlopt::opt::set_lower_bounds(class std::vector<double,class std::allocator<double> > const &)" (?set_lower_bounds@opt@nlopt@@QEAAXAEBV?$vector@NV?$allocator@N@std@@@std@@@Z)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_set_upper_bounds referenced in function "public: void __cdecl nlopt::opt::set_upper_bounds(class std::vector<double,class std::allocator<double> > const &)" (?set_upper_bounds@opt@nlopt@@QEAAXAEBV?$vector@NV?$allocator@N@std@@@std@@@Z)
sippynest2dpart0.cpp.obj : error LNK2019: unresolved external symbol nlopt_set_stopval referenced in function "protected: struct libnest2d::opt::Result<double> __cdecl libnest2d::opt::NloptOptimizer::optimize<class <lambda_2553785388a20e04dc430272b2ba8569> &,double>(class <lambda_2553785388a20e04dc430272b2ba8569> &,class std::tuple<double>,class libnest2d::opt::Bound<double>)" (??$optimize@AEAV<lambda_2553785388a20e04dc430272b2 [... cut ...]
E:\DEV\libnest2d\build\dependencies\lib\nlopt.lib : warning LNK4272: library machine type 'X86' conflicts with target machine type 'x64'
pynest2d.pyd : fatal error LNK1120: 32 unresolved externals
NMAKE : fatal error U1077: '"E:\DEV\software setups\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe"' : return code '0xffffffff'
Stop.
NMAKE : fatal error U1077: '"E:\Program Files\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"E:\Program Files\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.
```

Ghostkeeper, 2020-11-24:
Thanks. It looks like Clipper and NLopt are 32-bit binaries while you're building a 64-bit library.

Did you use the auto-download option of libnest2d to get those dependencies? You might need to give it some parameter to explicitly build 64-bit.

NoFish24, 2020-11-25:
I have the exact same issues with pynest2d. I have the auto-download option of libnest2d enabled and it gives the same error message as above.

Ghostkeeper, 2020-11-27:
While the instructions worked for our developers, they are still fairly new and probably not well fleshed out for various other systems.

One thing you could try is to not use the automatic download of these dependencies and install them manually. Or the reverse, try with `-DRP_FORCE_DOWNLOADING` instead.

konskarm, 2020-12-03:
Keep in mind that if you enable the `-DRP_FORCE_DOWNLOADING` flag, make sure you don't already have in your environment any of the dependencies, or else libnest2D won't be able to work well (as it will see it is installed and not download itself, but later it will search for that dependency in it's downloaded dependencies folder).

closed 2024-12-08 by github-actions[bot]