Ultimaker/Cura #8807: Support for right to left languages

AmrBarghouthi, 2020-11-20:
I want to add Arabic language to Cura the problem is it's right to left language which means the GUI needs to be flipped      
 
This would allow Cura to be translated to many right to left languages such as Arabic, Hebrew, Urdu and many more

PS: I am trying to work on this if some could point me to how correctly edit the GUI because I never used Qt before although I managed to run Cura from source

nallath, 2020-11-23:
I fear that's going to be a fair amount of work, since we (shame on us!) didn't really take this into account. So in quite a lot of cases we probably didnt' setup GUI elements in such a clean way that flipping them is possible.

That being said, we would be open to a PR that improves that. You will need to have a look at the QML files. I do know that qt has the option to set the text direction. It's a bit hard for me to figure out what needs to be done, since all languages that I know are left to right.

Ghostkeeper, 2020-11-24:
I'd think that it works okay for 90% of the texts. But some text fields are missing a width limit and those won't work correctly with RTL scripts.
As an example to get you started, take this label in the preferences page just left of the drop-down that selects the language:

https://github.com/Ultimaker/Cura/blob/39acad87d916b32c98e852b9e14b6b76b0070852/resources/qml/Preferences/GeneralPage.qml#L145-L149

Since it has no width, aligning to the right is not possible. I'd predict that this would make the text start from the left-hand side of the page and then proceed towards the left, overlapping with the menu of the preferences screen, and give off a QML warning in the log.
The fix would then be to give this a new `width` property, for instance a quarter of the total width of the preferences list:

    width: parent.width / 4

Possibly a better fix would be to give this the width of the text as measured by a TextMetrics object in this case (since it's currently using the GridLayout to align the drop-down to the right of the text).

Bigger problem is that I believe the norm is also to flip the whole workflow around, e.g. to put the drop-down to the left of that text. This sort of change will make this feature a lot of work, if it's deemed necessary.

AmrBarghouthi, 2020-11-26:
I don't mind that it going to be a lot of work after all I can take my time doin it
but I am still learning the QML syntax and I am wondering a do you guys use any tools to visuals the QML's or do i have to run the whole code every time i make change to see it

Ghostkeeper, 2020-11-30:
Some of us have used Qt Creator to visualise the QML more quickly. Not really to write the QML though, since Qt Creator doesn't integrate with the rest of our code base and is unable to handle the connections to Python code we put in there. The QML code is all hand-written.

Myself I basically start Cura every time. It takes 6 seconds to start on my computer now, so it's similar to compiling some C++ code; you get used to it.

nallath, 2020-12-01:
Ah! But there is a _secret_ key combination for this! Shift + f5 will trigger a reload of the qml files, without re-booting the interface! 

It won't work for all interface elements, since some of them are populated by plugins, but it should work for a fair number of them.

Ghostkeeper, 2020-12-03:
Yeah currently for me that refresh still crashes Cura with this error:

```
2020-12-03 11:44:12,700 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2020-12-03 11:44:12,700 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'dict' object to a C++ 'QVariantMap' instance
```

It's annoying that PyQt won't indicate any traceback there so it's really hard to figure out where in the code this happens.

AmrBarghouthi, 2020-12-03:
I have the same error as @Ghostkeeper 
anyway I am still learning QML syntax and Cura architecture in general 
hopefully by the end of the month I will make my first pull request

AmrBarghouthi, 2020-12-07:
@Ghostkeeper, @nallath I am defining a variable in the translation file to indicate the language direction, 
My question is what do you think it should be called I am thinking something like this

```
msgctxt "@language:direction"
msgid "ltr"
msgstr "rtl"
````
also where should it be defined in Cura project or the Uranium project or maybe both

Ghostkeeper, 2020-12-08:
I think that would be better suited as metadata field in the .po file header rather than a translated text.
Our translators are not involved in the development process and would get rather confused having to translate the text `ltr`. Maybe I'd translate that as l33tspeak for `later`?

AmrBarghouthi, 2020-12-08:
@Ghostkeeper you are right it is much better as metadata will update my pull request in the Uranium project in few minutes

AmrBarghouthi, 2020-12-08:
can someone explain why is this necessary 
https://github.com/Ultimaker/Cura/blob/bfe62514e2b8933df22ac361c88b5252ecfae5a2/resources/qml/LabelBar.qml#L45-L60
on my machine I got the same result just by
```qml
anchors.left: parent.left
```
which is easer to flip

nallath, 2020-12-09:
I think it had something to do with spacing on screens with a slightly different DPI. I'm not a 100% sure.

Usually, if there is a big wall of documentation, it's because it took an unrealisticly amount of time to fix it, and it's a sign that changing it should be done with care :P