Ultimaker/Cura #8808: Draft Shield flow rate

viper0078, 2020-11-20:
Sometimes the shield does not print cleanly.
[image]

The shield is not settling in the raft near the start of the shield printing.
I want to avoid making the raft airgap smaller because it makes it harder for the printouts to peel off.

It seems to me that it would be good to increase the flow rate of just the first layer of Drafut Shield.

viper0078, 2020-11-22:
There does not seem to be a draft shield in Layer0.
If it is a bug, this requirement is dropped.

Ghostkeeper, 2020-11-24:
> There does not seem to be a draft shield in Layer0.
If it is a bug, this requirement is dropped.

This is definitely a bug. I think it's assuming there is a brim or skirt beneath the draft shield, but you are using a raft. The code is checking for the `adhesion_type` setting and skipping that if it's using a brim. But apparently that check doesn't work properly?

The draft shield itself is a simple circle printed with the skirt/brim settings. It's using their line width, flow rate, speed, etc. It's a bit weird-ish, but the draft shield is such a detail that nobody would normally really care about its print speed and it should just use sane values.

Does this hole in your draft shield cause problems? It's only a few millimetres wide, so I'd expect that the effect of drafts is still extremely minimal. Fixing this can have a negative effect on productivity or print quality of your main print elsewhere.

viper0078, 2020-11-25:
On another printout, the printout was warped, starting at the point where the shield was punctured.
I don't want to waste my filament and time.

>Fixing this can have a negative effect on productivity or print quality of your main print elsewhere.
So you're saying that this bug is not a fix? That's a shame.

I created a conversion process that increased the flow rate to 200% for the first layer of the shield.
I have also increased the flow rate slightly after the second layer.
Now I can print consistently.

Ghostkeeper, 2020-11-26:
> Does that mean that this bug will not be fixed?
That's a shame.

No, I just meant that the goal of the draft shield is to protect against drafts. It doesn't necessarily need to be 100% tight and pretty to do that. You could increase the flow to try to fix this, at the cost of a big flow change when starting with the actual print, causing a blob there. So if there is no real reason to make the draft shield print watertight, then I'd rather not.

One concern that I'll raise is that the line width used for the draft shield is not visible if draft shield is enabled but skirt/brim is not. That doesn't address the issue that it's not at all clear that the Skirt/Brim Line Width is used for the draft shield, but I wouldn't really know how to fix that aside from creating a setting that almost nobody is going to change.

Ghostkeeper, 2020-11-27:
We've decided to make the Skirt/Brim settings visible if the draft shield is enabled for now.

But that leaves the problem that it's using those settings for the draft shield. This is both a UX problem (because those settings are not mentally connected at all to draft shields) and a printing problem (because those settings are probably optimised for printing the first layer). For that reason we're keeping this issue open.

viper0078, 2020-11-28:
Thank you. It's a nice improvement for me.

I think the sources for the draft shield should be copied from the skirt/blim sources and created separately.

closed 2024-12-08 by github-actions[bot]