Ultimaker/Cura #8809: PostProcessing plugin getSettingDataString is not interpreted

Sophist-UK, 2020-11-20:
**Application version**
4.8

**Platform**
Win 10 64-bit

**Printer**
Dagoma DiscoEasy200

**Reproduction steps**
I am writing a post processor script. I want to set defaults and maximums in getSettingDataString in the same way that can be done in fdmprinter.def.json e.g. `"minimum_value": "max(0, cool_fan_speed_min)",`.

Unfortunately plugins/PostProcessingPlugin/Script.py simply calls json.load and does not execute the Uranium Settings functionality that will interpret the python strings in the settings.

This makes it impossible for intelligent defaults / warnings / max/min settings to be defined in this JSON.

**Actual results**
If you try to use Python in e.g. the default_value attribute, a python error occurs.

**Expected results**
The python should be interpreted like the fdmprinter.def.json file is.

**Additional information**
I will try to work out how to achieve this and override the getSettingDataString function to test it out. If I can work out how to do this I will submit a PR for it.

Sophist-UK, 2020-11-20:
Sorry - but I have not worked out how to take the json from getSettingDataString and make it interpret the attributes in the same way as fdmprinter.def.json. I have been unable to find documentation to explain how Uranium helps and how Cura does this, nor discern this by reading code.

Sophist-UK, 2020-11-20:
If someone would like to provide me with an updated version of getSettingDataString from Script.py that does this, I would be happy to test it as an override in my development postprocessing scripts.

Sophist-UK, 2020-11-21:
Further research suggests that the PostProcessor plugin is using the right sort of object for formula in attributes, however I have still not been able to track down why it doesn't work. (Though I did submit PR https://github.com/Ultimaker/Uranium/pull/654 to fix the error I was receiving - just no idea why the same error is not received whilst processing fdmdetails.def.json.)

I have now found a way to simulate this in the `initialize` method, however setting `minimum_value_warning` with
```
self._instance.setProperty("minPrintSpeed", "minimum_value_warning", minPrintSpeed_default)
```
doesn't seem to work. This seems to be a separate issue, but again I am unable to track down why there is a problem.

Ghostkeeper, 2020-11-24:
It seems to work for the ChangeAtZ post-processing script. There it uses the `enabled` properties to make settings show or hide depending on other settings. And it has several minimum warnings and such.

You should be able to call `self.getSettingValueByKey` to get the evaluated setting value. Those properties like minimum value and warning values should get evaluated in the settings list to display the appropriate colours.

The JSON you provide in `getSettingDataString` gets put in a `DefinitionContainer` instance and deserialised there. This gets put in a container stack and setting values are evaluated from that container stack. A settings list is displayed on that container stack, similarly to how there's a settings list for each extruder stack. A lot of this works the same way as the normal settings list, really.

The features in the ChangeAtZ script seem to work fine. Could you compare your `getSettingDataString` with the one in there? Or else share it here so that other people can see what's wrong with it?

fieldOfView, 2020-11-24:
Note that as far as I remember, a postprocessing script cannot easily get a value from the global or extruder stack(s), only from the postprocessing script stack. This is why the PauseAtHeight script has a couple of hidden settings that get populated by the script backend before they can be used by in the setting definition:
https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/PauseAtHeight.py#L167
Gets set by the script here:
https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/PauseAtHeight.py#L213

Sophist-UK, 2020-11-24:
I will check both of these later this afternoon, but I have used existing scripts as the basis for my own, and I have tried using hidden settings.

Sophist-UK, 2020-11-24:
@Ghostkeeper :

1. PauseAtHeigth gets values from the global stack in both `initialize` and `execute` methods. I am unclear why it populates hidden settings in `initialize` when it seems to be able to get them just fine in `execute`.

2. PauseAtHeight uses `self._instance.setProperty` to save properties - I am unclear why it uses `_instance` because `getSettingValueByKey` uses `_stack`. I will have a play with this in the next day or so and see if using `_stack` fixes things.

Ghostkeeper, 2020-11-26:
A post-processing script gets a `ContainerStack` of profiles for itself, similar to an extruder train which gets a nozzle, material, quality profile, etc. of overrides stacked on top of each other. In the case of the post-processing script, this stack contains two profiles: The definitions that you write down there (which serve as the defaults) and a single setting instance container (`InstanceContainer`) that stores the changes that the user made.

As such, when you want to evaluate a setting, you need to do that from the `_stack`: It'll use the user-specified override if the user changed the setting, and if the user didn't change the setting it'll use the `"value"` you specified in the `getSettingDataString` function. If you want to store a new override programmatically, you'd need to do that in the `_instance` which is the profile containing user overrides.

Sophist-UK, 2020-11-26:
Got it - thanks for the explanation. However setting maximum_value_warning still doesn't work.

fieldOfView, 2020-11-26:
Post your full code somewhere, then we can have a look

Sophist-UK, 2020-11-26:
Sorry - I thought I had already done that.

https://github.com/Sophist-UK/Cura-PostProcessingScripts/tree/master/gCodePerSec

fieldOfView, 2020-11-26:
But there's no `maximum_value_warning` in there...

fieldOfView, 2020-11-26:
`minimum_value`, `minimum_value_warning`, `maximum_value` and `maximum_value_warning` all have to be Python-interpretable strings. So instead of `"minimum_value": 0` it should be `"minimum_value": "0"`

Sophist-UK, 2020-11-26:
Surely if they are NOT python interpretable strings they should instead be saved anyway???

So instead of e.g. `float(exec(setting))` it should be something like e.g. `float(exec(setting) if isinstance(setting, str) else setting)`.

However, I will try saving them as strings as see if that helps.

Sophist-UK, 2020-11-30:
Ok - I tried saving them as strings and it **didn't** help.
```
        if (prop.startswith("minimum") or prop.startswith("maximum")) \
          and not isinstance(value, str):
            Logger.log("d", "gCodePerSec: Converting " + key + "." + prop + " from " + str(type(value)) + 'to string: ""' + str(value) + '"')
            value = str(value)
```

fieldOfView, 2020-11-30:
I you have to make them strings mean that in your setting data string:
Eg [line 79](https://github.com/Sophist-UK/Cura-PostProcessingScripts/blob/61eaf8a/gCodePerSec/gCodePerSec.py#L79) `"minimum_value_warning": 10,`  should be `"minimum_value_warning": "10",`

I have no idea what you are trying to do with the code snippet above, and where you are putting it.

Sophist-UK, 2020-11-30:
> `minimum_value`, `minimum_value_warning`, `maximum_value` and `maximum_value_warning` all have to be Python-interpretable strings. So instead of `"minimum_value": 0` it should be `"minimum_value": "0"`

I have put it in the saveSettingProperty method in order to convert them to Python-interpretable strings as you suggested. So if it is not a string, it converts it to a string before trying to save it. However, regardless of using strings everywhere, I cannot manage to save e.g. minimum_value_warning successfully even though Uranium itself does not produce any error messages about the value not being saved.