Ultimaker/Cura #8810: Rotate uses original center when models are split apart using mesh tools

KenLucke, 2020-11-21:
**Application version**
4.8, but this has been ongoing (not a new issue, I just remembered it because it cropped up again in a project)

**Platform**
Mac OSX 15.5.7

**Printer**
N/A

**Reproduction steps**

1. Import a model capable of being broken into sub-models (the larger the separation between various parts, the more apparent will be the results)
2. Use "**Mesh Tools > Split Models Into Parts**" to split the model
3. Try to rotate one of new sub-parts.

**Screenshot(s)**

**Actual results**
The part will rotate around what was the _original_ center of of the whole model, rather than around its new individual center.  This also affects the "**Lay Flat**" and "**Select Face to Align to the Build Plate**" options.

**Expected results**
Each individual part (model) after a split should have its own individual center, just as if they had been imported individually.

**Project file**
N/A

**Log file**
N/A

**Additional information**

Ghostkeeper, 2020-11-26:
This has been reported before I think, but I can't find that issue now.

Something is going wrong in the way that the Mesh Tools plug-in creates the new meshes after splitting. But that usage of Cura's API should not go wrong, so it is considered to be a problem in Uranium itself rather than the plug-in.

fieldOfView, 2020-11-26:
> Something is going wrong in the way that the Mesh Tools plug-in creates the new meshes after splitting.

Actually it is doing exactly what it is designed to do. Mesh Tools keeps the original origin of the compound mesh, so it is possible to "merge" individual parts back together (eg after assigning them to different extruders) and having them "snap together" in their original position relative to eachother. If the Mesh Tools plugin were to reset the origin to the center of the sub-model, the position relative to the other sub-models would be lost and it would not longer be possible to merge them.

I'll add a function to move the origin to the center of a model in a future version of Mesh Tools, if @KenLucke would be so kind to open an issue at https://github.com/fieldofview/Cura-MeshTools/issues so I don't forget.

Ghostkeeper, 2020-11-27:
Indeed, and Cura should be able to deal with that as normal, but apparently something is going wrong there.

We've recently been dealing with a similar issue that was caused by the arrange functionality, so maybe this issue will get fixed by the same solution.

closed 2024-12-08 by github-actions[bot]