Ultimaker/Cura #8811: Radial Infill

GregValiant, 2020-11-21:
The concentric infill is kind of OK.  When the top layers go on they often fall through the cracks unless the infill is dense.  If there were radial lines involved then things like wheels would be more robust as the loads would be directed at the center.  In combination with concentric it would produce a grid.  I would much prefer it to the current concentric style infill.

**Describe the solution you'd like**
Concentric Infill.   Concentric Infill with Radial infill.   Radial infill.

**Describe alternatives you've considered**
Using a grid.

**Affected users and/or printers**
I expect the users would mostly be engineers and mechanical designers.  All printers affected.

**Additional context**
There would probably need to be minimum distancing or the radial infill would be extremely dense near the center, and maximum distancing to avoid being extremely sparse at the periphery.

alvgalrus, 2020-11-22:
I think this would be really useful for internal supports, where you want more density in the center (opposite to the "gradient infill").

Ghostkeeper, 2020-11-26:
Hmm, I could've sworn we've had this feature request a long time ago, with some drawings, but I can't find it now. There was this more recent incomplete issue: https://github.com/Ultimaker/Cura/issues/7405

There are 3 concerns that I'd wish to resolve before we can consider something like this:
* What is the use case of radiant infill? The current use case for concentric is (in my experience) only to prevent the infill from sticking to the walls which can create a shine-through effect, but this can also be achieved with the infill overlap setting. And maybe it could be used for flexible prints since it's the weakest infill of them all. But for radiant infill... I'd guess that radiant infill could be very resistant to pressure from the side, but it'd also be prone to buckling which is going to defeat that purpose largely. To get a nice top surface, lines or zigzag is a much better solution. (Or grid, as you proposed.)
* Where is the centre of the model? This can be quite obvious to a human when faced with a simple cylinder, but if Cura is presented with something [more complex](https://www.thingiverse.com/thing:1862365) then it still needs to produce an answer. This centre can't just be the middle of the bounding box of the layer, because that would then jump around all over the place between different layers causing the infill to hang in mid-air, and it would also be pretty much useless when printing anything that's not a convex shape. You'd almost need to create a skeletal graph of the model, but at that point things become fairly complicated.
* What would happen at the centre where all of the lines converge? From that previous feature request I remember some drawings that made the lines stop halfway towards the middle, causing fewer lines to reach the centre. However that has the disadvantage that it'll include more starts/stops and the lines that don't reach the centre will just... stop halfway somewhere, rendering them quite useless to improve strength unless printing at 100% density.

alvgalrus, 2020-11-30:
For the third question, just create a cilinder in the center, kind of be bottom shape in this image:
https://i.pinimg.com/originals/c2/0e/ae/c20eae05d58d438de89914dfba7d975e.jpg

This could be a complement to the concentric infill, adding radii from the innermost layer structure.

GregValiant, 2020-12-01:
Ghostkeeper,
I'll think on it some more and see if I can address some of the programming concerns.  From a mechanical perspective I can see such a feature as being very useful but maybe what I'm trying to accomplish is better left in the design stage than in the slicer.  I'll leave this open for now and see if @alvgalrus has anything to add.

Thanks,
Greg

pshunter, 2020-12-04:
My 2 cents if it helps : as far as I understand concentric infill prints several hulls, offset inwards. Instead of doing inner walls parallel to the outer shell, radiant infill should grow orthogonally from the shell. However, this is quite complicated since in the cylinder case some of these infill lines should stop before reaching the center so the density remains somewhat constant. In the concavous case, lines should start at a some distance from the shell to conserve density.
I believe doing concentric + radial might be easier, just interconnect inner infill walls with orthogonal lines, like in :
[image]

GregValiant, 2020-12-04:
I would find that acceptable.  In the case of cylinders, there could be a cutoff once a minimum distance was reached between the "spokes" of infill.  This is not an easy one to accomplish.  It's something I wanted for Christmas and I thought I'd ask Santa Cura for it.

Ghostkeeper, 2020-12-08:
I get the rough idea of that, but it has a problem.

There are examples where it's a bit ambiguous what should happen. In your drawing above you have corners that look a bit like this:
[image]
Note that the spoke density is not consistent on the inside nor the outside here. There are spokes on the inside very close together, but on the outside they are further apart than in the rest of the model too.

If you stretch that to a bit of a more extreme example, you get something like this:
[image]

But I'm wondering what should happen if it doesn't just stretch that corner straight up, but with a bend, like this:
[image]

We could make the spoke density consistent on the inside and let the spokes land on the outside wherever it intersects first. But that would then create a weak point in this bend. It's not an uncommon case either. You can see it starts to happen in Pshunter's drawing already in the innermost contour in the bottom right.

And the question remains what the use case of this infill is. Without a use case I see no reason to develop a new type of infill. Is it better in any way than the existing solutions? I'm seeing a lot of retractions here.

GregValiant, 2020-12-08:
The ellipses are closer to what I had in mind.  In the case of circles and ellipses any point on the exterior is "in view" of the center point of the radial infill.  In your Dr. Seuss model when the inside walls get out of "line-of-sight" of the picked center then yes, the idea falls apart.
I can see where it would not be any more advantageous to have radial + concentric as opposed to the grid + concentric.

Here is a view of concentric with Grid.  I think it helps to firm up the concentric infill and would provide structure to lay down a concentric top surface.

[image]

Ghostkeeper, 2020-12-17:
Yeah, it'd be more rigid than normal concentric that way. But not more rigid than just grid or triangular, right?

Luke2642, 2021-04-20:
What if we call this 'perpendicular struts' to augment concentric infill?

Whatever calculation is done the calculate concentric lines, it just needs flipping by 90 degrees every X mm, then extending until it touches the wall or another concentric infill line?

This solves the origin symmetry problem.

Ghostkeeper, 2021-05-06:
If you take the perpendicular line from the inside out rather than from outside in, and interrupt it on the first contour it hits going outwards, it wouldn't intersect any more. You'd have an uneven strength then if the object has any corners that are not radially symmetrical.

How would we make sure that the struts end up on the same place as struts on the previous layer as long as the shape is not discontinuous between layers?

Luke2642, 2021-05-06:
Correct Ghostkeeper, tradeoffs are expected. Uneven strength, uneven warping and uneven weight distribtuion are the current trade offs when using e.g. triangular infill with a circular object. In general, it's safe to say very few model geometries match any infill pattern to give even strength at low densities. 

To answer your strut placement question: The concentric lines placement algorithm copes with the changing perimeter between layers, and discontinuities. As long as a similarly 'stable' algorithm is used, with a small change in output for a small change in input, a small amount of bridiging will be an acceptable compromise.

gitcnd, 2021-07-27:
What's really needed for shapes like wheels is some kind of radial snowflake infill, so that the wheel loads are uniformly distributed between the inner of the rim and the outer of the tread.

Luke2642, 2021-07-31:
Specifying "maximum and minimum incident angle that infill lines can meet walls" could get us 80% of the way there with this feature and be simpler to implement. 

If a tight range was specified (say 85 to 95 degrees) using any normal infill pattern, then a "second pass" with the same infill pattern but rotated at e.g. 90 degrees would fill some the gaps. Then maybe a third pass at e.g. 45 degrees, etc? 

Alternatively, pseudocode for the patterns described could go something like this:

1. Draw normal concentric infill every C mm.
3. For each {outer wall or concentric infill line}
- pick any point on where infill met that line in the previous layer. If none exist, pick a random point.
- draw an 'radial' infill line at 90 degrees to the outer/concentric infill line
- When it collides with a wall or infill line, continue to draw an infill line alongside/slightly overlapping that line for T mm (creating @gitcnd's Y branches, with a bit of extra joining strength, like infill pattern multiplier). Stop if if any other line is encountered.
- pick another point on the outer wall X mm distance away, and repeat with the next in the for-each loop.
- if {no points on exist on each outer wall or each concentric infill line within X mm without a 'radial' line} : stop.

In some models determining the approximate centre of mass of the layer if it were 100% infill and then drawing towards that instead of 90 degrees could work, but that would be a 'nice to have' extra option.

emab123, 2022-04-06:
Hey guys, I'll be reviving this thread since I recently encountered a use case where such radial infill would be extremely useful.

I wanted to print a sort of pythagoras cup in which the external radius was determined by the size of the mount it was supposed to fit and the internal radius was determined by the volume it was supposed to hold. Therefore, the walls ended being somewhat thick on a big part, requiring a lot of infill. I only needed it to be stiff on the radial axis in order to maintain its volume, so I was very disappointed when I couldn't find any infill pattern that efficiently matched my load case. Take a look at this layer view:

[image]
In this case I wanted to make the inner wall as stiff as possible, so I would prefer an infill that creates lines perpendicular to the inner wall.  However I think there will be a lot of cases when someone wants to make the outer wall the stiffer, so it might be a good idea to offer the option of having either the inner of outer wall as the root of the radial line. Alternatively, a least material option in which each point of the the inner wall is connected to the closest point of the outer wall would eliminate any crossings of radial lines while offering somewhat of a midterm between inner and outer wall stiffness.

I believe that with these three settings most load cases will be covered if the radial wall stength is more important than than the layer adhesion strength. So the infill algorithm migh look a litte bit like this:

1. Draw a concentric infill every C mm (which influences the Z strength and radial line overhang) and determine it's print area.
2. Draw each radial line, based on the root wall setting and the infill percentage. This could be done somewhat like this:
> 1. Determine the root wall perimeter (p);
> 2. Determine the desired radial lines area by multiplying the total infill area by infill percentage and then subtracting the concentric infill area;
> 2. Choose the closest point on the root wall to (0,0) and draw a radial line according to the prefered setting, calculate it's area and add it to the current total area of the infill;
> 3. Choose the second point on the root wall, which is perimetically the furthest from the first point, so p/2, and draw a radial line from there and add it's area to the current infill area;
> 4. Choose the third and fourth root radial line points, which are the futhest from the first two with p/4, draw the radial lines and add their areas to the current infill area;
> 5. Keep drawing radial lines until the radial lines area is equal or more than what's desired. If p/2^n (where n is the loop counter) is less than the nozzle diameter, draw radial lines that start when the distance between the two adjacent radial lines is equal to the nozzle diameter.

daniels220, 2022-11-18:
@emab123 Your explanation in the now-closed CuraEngine#1635 is much more complete than what you have here—might you edit your comment and copy it over?

While I'm here, let me add another use-case for this pattern: supports. If the part being supported is relatively thin, mostly walls, the best support structure for it (easiest to remove relative to how effective it is as support) is going to be essentially a ladder or zigzag whose rungs are perpendicular to the walls it is supporting:

[image]

[image]

The "ladder" form would naturally come out of this proposal (with one support wall). The zigzag is actually better, though, so it might be better to approach it with that in mind as a separate proposal. Except, it's also important to have a pattern that is _just_ the perpendicular lines, no outer walls or zigzag at all, which I would use for the support _interface_, again for ease of removal (this is a top-down view—the top few layers would have only the purple lines, then the blue lines would appear once we get out of support interface):
 
[image]

emab123, 2022-11-18:
Splendid idea @daniels220 ! This should dramatically improve printing times for perimetric overhangs. However I feel like that would be a different issue. What do you think about moving this idea to a new issue so we can discuss specifically the infill around here?