Ultimaker/Cura #8812: [4.8.0] It takes too much time (up to few minutes) to load the object (*.stl) or project (*.3mf)

MHB8080, 2020-11-21:
**Application version**
4.8.0 (some older versions I've tested are affected too)

**Platform**
Linux, Ubuntu 18.4, 64-bit

**Printer**
Ender 3

**Reproduction steps**
1. Start Cura
2. Import/Open any *.stl file

**Actual results**
It takes too much time  (minutes?) to load the object/project

**Expected results**
The object/project should be loaded without wait time of tens of seconds

**Additional information**
Black box observation: It seems to me the loading is possible only after all the "USBPrinting.AutoDetectBaudJob.run [50]: Checking /dev/ttyACM0" checks/logs are over. Previously, my user account had no access rights to these /dev/tty* devices, but few days ago I granted the rights to them for my user account because I had to use them, inclusive some /dev/ttyUSBx devices. But, as already mentioned, it is pure black box observation/feeling.

nallath, 2020-11-23:
On linux we use a fallback for the STL loading, since we got some crashes with the faster solution that we had. 

Your reproduction steps don't mention issues with 3mf. Is this intentional? If the issue exist with both 3mf and STL, it's a slightly more fundamental thing that is going on.

Alternatively; Does the issue persist if you disable the USBPrinting plugin (you can do this in marketplace -> installed)

MHB8080, 2020-11-24:
1. The issue was the same (from my POV, at least) for both *.stl AND *.3mf
2. After disabling the USBPrinting plugin (as you mention) fixed the issue for both *.stl and *.3mf
3. After the USBPrinting plugin was REenabled -> issue is back again
4. After the USBPrinting plugin was DISabled (same as in step 2) then everything is OK from loading of *.stl and *3mf POV
5. Thanks a LOT! :-) Because I don't need the USBPrinting plugin -> I am happy now :-D

Ghostkeeper, 2020-11-26:
So then the issue is really with that the USB printing auto-detection of USB-connected printers takes very long. Other asynchronous jobs then get put in the queue (a decision made due to the GIL of CPython).

Ghostkeeper, 2020-11-26:
Actually, this is really a duplicate of https://github.com/Ultimaker/Cura/issues/5600.

closed 2020-11-26 by Ghostkeeper