Ultimaker/Cura #8813: Here is why you should add a "minimum extrusion length" setting

RedPetroleum, 2020-11-21:
**Application version**
(all versions) I'm using 4.8.0

**Platform**
MacOS 

**Printer**
Custom FDM
**Screenshot(s)**
[image]

**Actual results**
[image]

**Additional information**
Tiny inner wall extrusions can be visible on the surface. The settings _Fill gaps between walls_ and _Filter out tiny gaps_ are handy but don't solve this particular issue. It would be nice to have a setting that just filters out all tiny extrusions.

RedPetroleum, 2020-11-21:
This one is also pretty obvious. 

[image]

Ghostkeeper, 2020-11-26:
Yet these lines are printed properly. Basically it's going to be printing dots in there. Those dots are quite useful. Here's a print with similar dots:

[image]

When you actually print that, those dots fill up the gaps in the walls quite nicely. Here's a comparison without those dots on the left, and with those dots on the right:

[image]

This is really because the nozzle goes over that spot, stands still there for a fraction of a second extruding material while realising that it doesn't have time to accelerate and decelerate, and then moves on. I'd argue that this is a good thing. The line may be very short (and thus visualised almost invisibly in the legacy layer view) but it actually prints fine.

Is there some advantage you could name? It's going to save some time to not print this, but that won't be very significant save in some extreme cases.