Ultimaker/Cura #8814: auto-generated code before start gcode

AkahitoTheDragon, 2020-11-21:
**Application version**
4.8.0

**Platform**
Windows 10

**Printer**
Anycubic I3 Mega

**Reproduction steps**
1. sliced a project
2. save gcode

**Actual results**
Autogenerated Code before Start-Code handles heatup and makes the start code superfluous

**Expected results**
Only Start-Gcode handle the temperatures etc.

**Log file**
2020-11-21 22:18:19,641 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-21 22:18:19,644 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G21 ;metric values
2020-11-21 22:18:19,648 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G90 ;absolute positioning
2020-11-21 22:18:19,652 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] M82 ;extruder absolute mode
2020-11-21 22:18:19,656 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] M107 ;fan off
2020-11-21 22:18:19,659 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-21 22:18:19,662 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G92 E0 ;reset extruder to 0
2020-11-21 22:18:19,665 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G28 X Y F1800 ;Home X & Y
2020-11-21 22:18:19,668 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] M190 S50 T0 ;heat bed
2020-11-21 22:18:19,671 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G28 Z F1800 ;Home Z
2020-11-21 22:18:19,674 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] M420 S1 ;Enable ABL using saved Mesh
2020-11-21 22:18:19,677 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-21 22:18:19,680 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 X-5 Y10 F1800 ;move to prime position
2020-11-21 22:18:19,683 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 Z0.5 ;move up
2020-11-21 22:18:19,686 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] M109 S195 T0 ;heat extruder
2020-11-21 22:18:19,689 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 E20 Z3 F20 ;prime nozzle
2020-11-21 22:18:19,692 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 E30 Z4 F10 ;prime nozzle
2020-11-21 22:18:19,695 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 Z10 F1800 ;move up
2020-11-21 22:18:19,699 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 X70 Y3 Z0.1 F1800 ;move to prime line start
2020-11-21 22:18:19,702 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G92 E0 ;zero the extruder
2020-11-21 22:18:19,705 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 X165 Y3 E8 F600 ;purge nozzle
2020-11-21 22:18:19,708 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 Z2.0 F3000 ;Move Z 

**Additional information**
Start-Gcode:

M82 ;absolute extrusion mode

G21 ;metric values
G90 ;absolute positioning
M82 ;extruder absolute mode
M107 ;fan off

G92 E0 ;reset extruder to 0
G28 X Y F1800 ;Home X & Y
M190 S{print_bed_temperature} T0 ;heat bed
G28 Z F1800 ;Home Z
M420 S1 ;Enable ABL using saved Mesh

G1 X-5 Y10 F1800 ;move to prime position
G1 Z0.5 ;move up
M109 S{print_temperature} T0 ;heat extruder
G1 E20 Z3 F20 ;prime nozzle
G1 E30 Z4 F10 ;prime nozzle
G1 Z10 F1800 ;move up
G1 X70 Y3 Z0.1 F1800 ;move to prime line start
G92 E0 ;zero the extruder
G1 X165 Y3 E8 F600 ;purge nozzle
G1 Z2.0 F3000 ;Move Z Axis up

nallath, 2020-11-23:
Where did you place that start-gcode? In the extruder or in the machine?

AkahitoTheDragon, 2020-11-23:
in the "printer" tab

fieldOfView, 2020-11-23:
You should be using `{material_print_temperature_layer_0}` instead of `{print_temperature}`. If you do so, Cura will not have to prepend its own preheat sequence. If your printer has a heated bed, you also have to add preheating with {material_bed_temperature_layer_0} to your start gcode.

fieldOfView, 2020-11-23:
>  The "Printing Temperature Initial Layer" is also set to 0 in the settings.

Well, then don't set it to 0 in the settings.
Alternatively, you can use `{material_print_temperature}` instead of `{material_print_temperature_layer_0}`

AkahitoTheDragon, 2020-11-23:
the default setting is 0, `{material_bed_temperature_layer_0}` default setting is also 0
but `{material_bed_temperature_layer_0}` take the normal bed temp, {material_print_temperature} not.

using `{material_print_temperature}` creates the preheat sequence again.

fieldOfView, 2020-11-23:
> the default setting is 0

No it isn't. The default behavior is to copy the material_print_temperature:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json#L2189

> using {material_print_temperature} creates the preheat sequence again.

I cannot reprodice that.

AkahitoTheDragon, 2020-11-23:
It seems in the standard profile of the "Anycubic I3 Mega" the standard value is 0, not the calculated one from the `{material_print_temperature}`. Ive corrected that.
Now `{material_print_temperature_layer_0}` and `{material_bed_temperature_layer_0}` gives me the right values now.
Problem solved, thank you!

closed 2020-11-23 by AkahitoTheDragon