Ultimaker/Cura #8815: No draft shield in Layer 0

viper0078, 2020-11-22:
There is no draft shield in Layer0.
The creation of a draft shield fails because a draft shield is printed from Layer 1.
******
;LAYER:1
M204 S800
G92 E716.78826
;TYPE:SKIRT
G1 F3600 E718.78826
G1 F1080 X144.433 Y166.544 E718.81983
******
[image]

[image]

viper0078, 2020-11-23:
It seems to be a bug that occurs when using rafts.

Ghostkeeper, 2020-11-26:
(Also reported in #8808, but rightly split off to a separate ticket. Thank you.)

Indeed this issue is easily reproduced. I've put this on the list of new issues to discuss with the rest of the team.

Ghostkeeper, 2020-11-30:
(Developers, refer to issue CURA-7894.)

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-24 by github-actions[bot]