Ultimaker/Cura #8816: Abort Print on anycubic i3 mega

xpatchdev, 2020-11-22:
**Application version**
all
**Platform**
linux

**Printer**
anycubic i3 mega

**Reproduction steps**
abort a print
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
the printer bed moves backwards, but when a print completes normally it rolls forwards

**Expected results**
print bed should roll forwards and should jump up in the z coordinate a bit just in case the print head got jammed on the printer bed. instead I click abort click up on Z and then I mash on the down Y because it rolls backwards fully before I can control it again

**Project file**
any file

**Log file**
O.o should I wireshark the USB data?

**Additional information**
I'm so tired I only just noticed this is github... I could prob fix myself lol.

nallath, 2020-11-23:
Cura doesn't have a concept of aborting when printing over USB other than just "stop sending data". 

The USB printing hasn't received any love from us and is not likely to get it either since none of our printers use it. But if you feel like it, feel free to make a PR to improve it for people that do still use it :)

closed 2020-11-26 by nallath

xpatchdev, 2020-11-29:
So you guys only print by exporting gcode and loading it into the printer?

xpatchdev, 2020-11-29:
Could you point me to the place in src where I might insert onAbort( if
printer == anycubic i3 mega { move(+Z 10); move(+Y 80)) type of
instructions.
Is there a place to put gcode cmds based on success or abort maybe in
printer settings.
when the printer finishes normally it rolls the bed forward but on abort it
rolls backwards, this might be a manufacturing
thing where during automation if the bed moves backwards the print is bad
and discarded or forwards and it's picked up by
the smiling robot arm for further processing.
I'll figure it out, but if you know off hand