Ultimaker/Cura #8817: [4.7.1] AuthorsModel.py: error: Argument "key" to "sort" of "list" has incompatible type

onitake, 2020-11-22:
**Application version**
4.7.1
Still present in latest HEAD (39acad87d)

**Platform**
N/A

**Printer**
N/A

**Reproduction steps**
1. Run the mypy code-style test (`python3 run_mypy.py`)

**Screenshot(s)**
N/A

**Actual results**
The test fails with the following error:
```
plugins/Toolbox/src/AuthorsModel.py:71: error: Argument "key" to "sort" of "list" has incompatible type "Callable[[Dict[str, Union[str, List[str], int, None]]], Union[str, List[str], int, None]]"; expected "Callable[[Dict[str, Union[str, List[str], int, None]]], _SupportsLessThan]"
plugins/Toolbox/src/AuthorsModel.py:71: error: Incompatible return value type (got "Union[str, List[str], int, None]", expected "_SupportsLessThan")
Found 2 errors in 1 file (checked 17 source files)

Command /usr/bin/python3 /usr/bin/mypy -p Toolbox --ignore-missing-imports failed checking. :(
```
All other tests are successful.

**Expected results**
No error is reported.

**Project file**
N/A

**Log file**
N/A

**Additional information**
This is causing fallout for Debian packaging right now, because running the unit tests is part of the build process:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=973230

Please provide a fix urgently or advise if we should disable the mypy test for the time being.
Thank you.

nallath, 2020-11-23:
Uuh. That's weird. We run the mypy tests as well with every commit. Right now they are not failing.

onitake, 2020-11-23:
I just found it's a 4.8 issue, not 4.7.1.
My system has all the 4.8 frameworks (Savitar, Uranium, Arcus) installed, but the package build of Cura 4.8 fails due to the autopkgtest.

What's the configuration of the commit test? Maybe we can pin it down like that.

nallath, 2020-11-23:
We run it with mypy 0.740

onitake, 2020-11-26:
That could be the problem.
Debian testing/stable has python3-mypy: 0.790-2

Could you try to reproduce using this version, please?

nallath, 2020-11-27:
I've tried running it with 0.780 and 0.790, but both don't work at all. Mypy itself fails with an exception:
`AttributeError: module 'mypy_extensions' has no attribute 'mypyc_attr'`

I've not been able to find a lot of helpful information on how to solve that. Switching back to 0.740 fixes the issue.

onitake, 2020-11-27:
I have python3-mypy-extensions 0.4.3-2 installed - did you update that as well?

I'm afraid that we can't run tests with an older version of mypy - the only other option would be to disable the test.
But I'd really prefer if you could investigate the error with mypy 0.790 now. An upgrade will be inevitable at some point anyway, and the mypy error message suggests it could be a real problem.

nallath, 2020-11-27:
> I have python3-mypy-extensions 0.4.3-2 installed - did you update that as well?

I have 0.4.3 installed. 

Anyhow, I could try and fix this without it, but it might take a few shots to see if it worked.

AdrianBunk, 2020-12-05:
The build problem of 4.7.1 in Debian is fixed with commit 00bc1bfcdc960d4bad62adc4b1615ca4cc6b3cee.

onitake, 2020-12-05:
@AdrianBunk Oh sorry, I didn't see a fix was committed.

@nallath Thanks for the patch. I'll push it to the Debian repo immediately.

nallath, 2020-12-07:
This can be closed, right?

onitake, 2020-12-07:
If you're ok with the cast as a solution.

I looks more like a workaround to me, but I don't understand why the test failed in the first place, so I don't rightly know.

nallath, 2020-12-07:
I don't either. But the cast is there for exactly that reason.

closed 2020-12-07 by nallath