Ultimaker/Cura #8819: I can't monitor my ANET A6 with USB connection

Chrispizz, 2020-11-22:
**Application version**
CURA 4.8.0

**Platform**
WINDOWS 10

**Printer**
ANET A6

**Reproduction steps**
Connect ANET A6 from USB 
Nothing appears on MONITOR tab

**Screenshot(s)**
Nothing to show , just MONITOR tab empty 
[image]

**Actual results**
Nothing happens , just MONITOR tab empty 

**Expected results**
My ANET A6 is connected to USB ( COM5 ) then MONITOR tab should let me monitor the ANET A6 with controls buttons on the screen 

**Project file**
No project to show 

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5580075/cura.log)

**Additional information**
I hope you will tell me how I can do it work

nallath, 2020-11-23:
You didn't give Cura permission to actually read data over the USB port.

It's stating that an "Accès refusé." exception happend.

closed 2020-11-23 by nallath

Chrispizz, 2020-11-23:
I've just added the diver CH340G for windows 10
I see more good logs for still not works , MONITOR tab is still empty without control buttons

[cura.log](https://github.com/Ultimaker/Cura/files/5582630/cura.log)

nallath, 2020-11-23:
You still have the same error; Cura doesn't get the permission to read the serial.