Ultimaker/Cura #8820: After update all data seems to be gone and stuck in the wizard

forenbuddy, 2020-11-22:
**Application version**
(The version of the application this issue occurs with.)
4.7.1-2
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Debian Linux Bullseye

**Printer**
(Which printer was selected in Cura?)
None. It is stuck at the Wizard. It asks for Printer name but no input brings any action.

**Reproduction steps**
1. (Something you did.)
Updated my machine.
Started the program. 

2. (Something you did next.)
Cura was working fine until the update. I printed many objects. Now forcefully had to go through the wizard until it asks to Add a printer. It now asks for Printer name but no input brings any action. I can only close the program.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)
There is the input field labeled Printer name but whatever I enter nothing happens any more. The program is stuck.

**Expected results**
(What should happen after the above steps have been followed.)
I should be able to choose my printer I guess or at least be able to press cancel or something to use the program.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

/usr/lib/python3/dist-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
[MainThread] UM.Application.initialize [144]: Initializing Ultimaker Cura
[MainThread] UM.Application.initialize [145]: App Version 4.7.1
[MainThread] UM.Application.initialize [146]: Api Version 7.3.0
[MainThread] UM.Application.initialize [147]: Build type None
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/bin/share/uranium/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/bin/share/cura/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/lib64/uranium must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/lib32/uranium must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/bin/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/Resources/uranium/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/Resources/cura/plugins must be a folder.
[MainThread] UM.Resources.__initializeStoragePaths [446]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [456]: Config storage path is /home/x/.config/cura/4.7
[MainThread] UM.Resources.__initializeStoragePaths [464]: Data storage path is /home/x/.local/share/cura/4.7
[MainThread] UM.Resources.__initializeStoragePaths [476]: Cache storage path is /home/x/.cache/cura/4.7
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/lib/python3/dist-packages/UM/../plugins must be a folder.
[MainThread] UM.PackageManager.__init__ [54]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager.__init__ [54]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [118]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [126]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [173]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [180]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [184]: Initializing version upgrade manager ...
/usr/lib/python3/dist-packages/cura/UI/CuraSplashScreen.py:73: DeprecationWarning: an integer is required (got type float).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
  painter.drawText(60, 70 + self._version_y_offset, 330 * self._scale, 230 * self._scale, Qt.AlignLeft | Qt.AlignTop, version[0])
[MainThread] UM.PackageManager._loadManagementData [165]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [165]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [186]: Loaded user packages management file from /home/x/.local/share/cura/4.7/packages.json
[MainThread] UM.PackageManager._saveManagementData [259]: Package management file /home/x/.local/share/cura/4.7/packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [115]: Loading plugin configuration file '/home/x/.config/cura/4.7/plugins.json'
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/lib32/cura must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/lib64/cura must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/libx32/cura must be a folder.
[... 352 lines cut ...]
[MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/usr/share/cura/resources/shaders/transparent_object.shader]...
[MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/usr/share/cura/resources/shaders/striped.shader]...
[MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/usr/share/cura/resources/shaders/xray.shader]...
[MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/usr/share/cura/resources/shaders/xray_composite.shader]...
[MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/usr/share/uranium/resources/shaders/platform.shader]...
/usr/lib/python3/dist-packages/UM/View/GL/FrameBufferObject.py:18: DeprecationWarning: an integer is required (got type float).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
  self._fbo = QOpenGLFramebufferObject(width, height, buffer_format)
/usr/lib/python3/dist-packages/UM/View/GL/ShaderProgram.py:336: DeprecationWarning: an integer is required (got type float).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
  QColor(value.r * 255, value.g * 255, value.b * 255, value.a * 255))
[MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/usr/share/cura/resources/shaders/xray.shader]...
[MainThread] cura.CuraApplication.writeToLog [595]: User accepted the User-Agreement.
[MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
[MainThread] cura.OAuth2.AuthorizationService.startAuthorizationFlow [161]: Starting new OAuth2 flow...
[MainThread] cura.OAuth2.LocalAuthorizationServer.start [63]: Starting local web server to handle authorization callback on port 32118
[MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
[MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///usr/share/cura/resources/qml/WelcomePages/DropDownWidget.qml:35:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///usr/share/cura/resources/qml/WelcomePages/DropDownWidget.qml:35:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///usr/share/cura/resources/qml/WelcomePages/AddLocalPrinterScrollView.qml:213:17: Unable to assign [undefined] to QString
[MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///usr/share/cura/resources/qml/WelcomePages/AddLocalPrinterScrollView.qml:237:21: Unable to assign [undefined] to QString
[MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///usr/share/cura/resources/qml/WelcomePages/AddLocalPrinterScrollView.qml:254:21: Unable to assign [undefined] to QString
[MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
[MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-11-23:
Well, you clearly didn't run the appimage. It's missing a lot of dependencies and it's using the wrong versions of a whole bunch of plugins.

So the solution here would be to download (and use) the appimage.

closed 2020-11-23 by nallath

darkpoetry, 2020-11-26:
If UM isn't going to support Cura on Debian, it should notify Debian and pull it out of the package repositories. Otherwise it should get updated with the appropriate dependency requirements. Bug should probably be re-filed in Debian Bullseye.

Ghostkeeper, 2020-11-30:
We do support Debian, through the AppImage.

We are not the boss over package maintainers though. We are not going to forbid those people from re-packaging Cura for various Linux distributions. They are free to redistribute Cura as they please, per our license agreement (with some restrictions e.g. to publish any modifications they make to our source code). That they are not always doing that successfully is then not really our concern. We like to help if package maintainers have questions, but making Cura work with future breakages in Qt is out of scope for us.

I think they should make Cura depend on a specific major+minor version of these dependencies.

forenbuddy, 2020-12-03:
@Ghostkeeper The main reason for using a distribution repository is to only have to trust this distribution and its compile mechanisms and not have to trust also every single upstream package maintainer to ship the packages really compiled from the sources and not something (additionally) else.

So when you say you support Debian via AppImages you really are not supporting the Debian project but only some Debian users who don't care if their system may be tainted with packages from potentially unsafe sources.

soern, 2020-12-13:
The whole point of Distributions and Package repositories is to have a) a trusted source, b) to keep all the dependencies aligned and c) to use shared libraries to avoid bloat. Appimage is a "fix", but it violates principle c. If you'd now do this for every application, imagine how long it'd take to update applications, how many system resource you'd use and how many redundant copies and copies libraries you'd have for all those 'unicorn' applications...

I understand the reasons on both sides and I'd like to suggest a resolution that I think would make everyone happy. I believe that part of the reason is, that cura developers were under the assumption that all their users use appimage to automagically resolve dependencies. How about clearly naming these dependencies so that packet maintainers may pick them up?

fieldOfView, 2020-12-13:
> How about clearly naming these dependencies so that packet maintainers may pick them up?

Like in this file?
https://github.com/Ultimaker/Cura/blob/master/requirements.txt

Ghostkeeper, 2020-12-18:
That file only names the Python dependencies. It's also a (partial) duplication of the actual dependencies listed in the [cura-build-environment](https://github.com/Ultimaker/cura-build-environment/tree/master/projects) repository.

It's fine to make Cura share dependencies with other applications, of course. As long as they are the same dependencies, and not newer versions of those dependencies that have breaking changes.