Ultimaker/Cura #8822: 4.8's improved "arrange" algorithm places models colliding with each other and overlapping

thanhtacles, 2020-11-23:
Both dragging in multiple models, and multiplying models causes them to collide and overlap with each other. Pretty obvious unintended use, as this will cause a ton more time spend rearranging objects.

Thank you for using Cura!
-->

**Application version**
4.8

**Reproduction steps**
drag or multiply any model at all. They will collide no matter what. 

**Screenshot(s)**
[image]

**Expected results**
Models used to be spaced part pretty cleanly, even though they were of the same orientation.

nallath, 2020-11-23:
Does this happen for all models with you? I've not been able to reproduce this (eg; All models that I load get nicely spread over the build plate).

If it's just these models, could you share them with us?

thanhtacles, 2020-11-23:
No it's all models

Vandrasc, 2020-11-23:
@nallath is this not a duplicate of https://github.com/Ultimaker/Cura/issues/8779?

nallath, 2020-11-23:
Yeah, I think so. But I am a bit suprised how / why this happens for all models.

Ghostkeeper, 2020-11-26:
It doesn't happen with models where the bounding box is square (as opposed to rectangular).

A workaround is to enable "ensure models are kept apart" in the preferences.

linuxdan, 2021-03-10:
Good workaround, but why isn't that "feature" auto-applied to auto arrangement ?

Ghostkeeper, 2021-03-23:
Closing this as duplicate of https://github.com/Ultimaker/Cura/issues/8779.

closed 2021-03-23 by Ghostkeeper