Ultimaker/Cura #8823: Problems with cura 4.8

yzorgone, 2020-11-24:
I am on a new Macbookpro (2019) With Big Sur OS.

Cura release. 4.8

Ultimaker S5

Problem 1.
after slicing layer preview is not visible anymore. does not come up, just dark viewport. layer slider are there but no use.

Problem 2.
Cura operator camera zoom zooms to an irrelevant point in the left corner of the printer.
When Selecting the object the camera resets view to center. but the zoom point stays offset to that left corner. This results in: when zoomed in, the Camera rotates not around the desired object anymore.

Zoom problem video:
https://youtu.be/LcYjNyKO2lA

jellespijker, 2020-11-26:
Please use the bug template so we have all relevant information and create separate reports for each problem. This will make administration manageable for us.

Ghostkeeper, 2020-11-27:
Duplicate of https://github.com/Ultimaker/Cura/issues/8784

closed 2020-11-27 by Ghostkeeper