Ultimaker/Cura #8828: Reading STL from Smoothie-3d into Cura v4.8

JJRacoon, 2020-11-25:
When opening an STL file created in Smoothie-3d in Cura Ver 4.8 all that is processed is a 2d image of the footprint on the buildplate.  Works in v4,7

**Application version**
4.8

**Platform**
Windows 10

**Printer**
CR10s

**Reproduction steps**
1. Opened file in Cura v4.7 -success - 3d
2. Opened file in Cura 4.8 - failure - 2d

**Screenshot(s)**
Version 4.7:
[image]

Version 4.8:
[image]

**Actual results**
See screenshot "On v4.8" above

**Expected results**
See screenshot "On v4.7" above

**Project file**
[Pot from Smoothie3d.zip](https://github.com/Ultimaker/Cura/files/5593474/Pot.from.Smoothie3d.zip)

nallath, 2020-11-25:
This is a known issue; It doesn't seem to correctly drop the model to the buildplate. It's placing the object very high above the build plate. 

You can either rotate the camera to look up, or select the model from the object list and then set the z to be zero.

closed 2020-11-25 by nallath

JJRacoon, 2020-11-25:
Thanks nallath

Ghostkeeper, 2020-11-27:
(See https://github.com/Ultimaker/Cura/issues/8752.)