Ultimaker/Cura #8831: When opening second project of session, Cura does not allow object selection

RealLibrarian, 2020-11-25:
**Application version**
Cura 4.8.0

**Platform**
MacBook Pro (2016) Big Sur

**Printer**
Creality Ender 5 Plus

**Reproduction steps**
1. Open .stl file, slice, save project.
2. File > New Project
3. Open new .stl file

**Screenshot(s)**

**Actual results**
Cannot select object to manipulate or change in this session

**Expected results**
Should be allowed to select and manipulate

**Project file**
Works for any .stl

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5599278/cura.log)

(Extra information relevant to the issue.)
Exit and restart of Cura fixes problem.

nallath, 2020-11-26:
Possibly related to https://github.com/Ultimaker/Cura/issues/8745

Ghostkeeper, 2020-12-28:
I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/8745, indeed. There are multiple symptoms of that issue, because when you're unable to interact with the 3D scene that also means you can't rotate objects with the rotation handles, scale objects with the scale handles, select objects, etc.

closed 2020-12-28 by Ghostkeeper