Ultimaker/Cura #8832: Tree supports often start too high

mereseydotes, 2020-11-25:
**Application version**
4.8.0

**Platform**
Windows 10

**Printer**
CR6- SE

**Reproduction steps**
1. Opened model and sliced with tree supports

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Sometimes, when a tree support has a wider base, it requires additional folds (?) for adequate support. Sometimes these folds don't start printing until sometime up the model, which means they print in the air and "spaghetti" until they take hold. I will be attaching a file that has a support like this at the back of the model. The "bad" supports tend to grow on the inside of existing tree supports.

**Expected results**
No spaghetti

**Project file**
Link: https://drive.google.com/file/d/1r4bIib9AfLyF-NVtpBHAK2IoNa1Yo6g-/view?usp=sharing

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mereseydotes, 2020-11-26:
I have also noticed that some models do not get adequate support with support on buildplate only.

Ghostkeeper, 2020-11-27:
Duplicate of https://github.com/Ultimaker/Cura/issues/4342.

closed 2020-11-27 by Ghostkeeper