Ultimaker/Cura #8835: zits with 4.8 but not 4.6.2 on the same profile

Spacecowboy83, 2020-11-26:
I have an FLSUN Q5 with the original firmware. Zits appear in 4.8 but not 4.6.2 on features like outside sharp curves. Same profile where the resolution settings are set manually to the 4.6.2 defaults. Attached are screenshots of the issue with 4.8 and 4.6.2 for comparison, the profile and the two G-codes which were compiled with 4.6.2 and 4.8 respectively

[image]
[image]

[Project + G-codes.zip](https://github.com/Ultimaker/Cura/files/5602082/Project.%2B.G-codes.zip)

nallath, 2020-11-26:
You shouldn't set the resolution to the same value as in 4.6.2. That version had a bug in it that wasn't reducing the resolution correctly. 

Check the discussion in https://github.com/Ultimaker/Cura/issues/8321 for more information.

closed 2020-11-26 by nallath

Spacecowboy83, 2020-11-26:
I also tried other prints before with the default resolution and deviation settings of 4.8. And these produced the same issue. 4.6.2 always worked perfectly for me - 4.8 is the version which produces the blobs - not version 4.6.2

Spacecowboy83, 2020-11-26:
Also this is not a duplicate since this is about version 4.8, not version 4.7

nallath, 2020-11-26:
Have you tried making the resolution even more coarse? Because that is what 4.6.2 was actually doing.

Spacecowboy83, 2020-11-26:
Thanks for the answer! Is that the value for max resolution or something else and should I increase or decrease it and which increments would you suggest?

nallath, 2020-11-26:
You need to increase that, same for the "Maximum Deviation". I'd try to double it first and see what effect that has.

Spacecowboy83, 2020-11-26:
Alright, thanks, I will try that within the next few days when I print models which are suited for testing.

Spacecowboy83, 2020-12-02:
> You need to increase that, same for the "Maximum Deviation". I'd try to double it first and see what effect that has.

I tried different resolution and deviation values of up to 0.2 for resolution and 0.3 for deviation, now and I still get the same pattern of blobs in the same model. Still clean in version 4.6.2. I believe this issue has nothing to do with resolution after all and might be a new bug.