Ultimaker/Cura #8836: Cura 4.8.0 randomly freezing / not responding after update

Beerlord666, 2020-11-26:
**Application version**
4.8.0

**Platform**
OS: Windows 10 Pro 2004
RAM: 16 gb
CPU: i5-7600K
GPU: Nvidia GTX1070

**Printer**
Anycubic i3 Mega

**Reproduction steps**
1. start cura - takes abozut 20-30 seconds for some reason
2. do anything in cura (eg. load a .stl file) and view it from a few sides OR open marketplace

**Screenshot(s)**
---

**Actual results**
cura freezes for 10-20 seconds, then you can continue working with it for up to 1 minute until it freezes again

**Expected results**
cura does not freeze

**Project file**
--- no project file needed ---

**Log file**
2020-11-26 15:06:24,315 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ConsoleLogger
2020-11-26 15:06:24,347 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraDrive
2020-11-26 15:06:24,362 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [92]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.8.0\CuraEngine.exe
2020-11-26 15:06:24,362 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraEngineBackend
2020-11-26 15:06:24,378 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileReader
2020-11-26 15:06:24,378 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2020-11-26 15:06:24,378 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: C:\Program Files\Ultimaker Cura 4.8.0\CuraEngine.exe
2020-11-26 15:06:24,393 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.8.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2020-11-26 15:06:24,393 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileWriter
2020-11-26 15:06:24,409 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJReader
2020-11-26 15:06:24,416 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJWriter
2020-11-26 15:06:24,416 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-26 15:06:24,416 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.8.0
2020-11-26 15:06:24,431 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2020-11-26 15:06:24,431 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-26 15:06:24,431 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2020-11-26 15:06:24,431 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-11-26 15:06:24,447 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-11-26 15:06:24,447 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2020-11-26 15:06:24,447 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-26 15:06:24,447 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2020-11-26 15:06:24,462 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLReader
2020-11-26 15:06:24,462 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2020-11-26 15:06:24,462 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2020-11-26 15:06:24,462 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2020-11-26 15:06:24,462 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2020-11-26 15:06:24,462 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2020-11-26 15:06:24,478 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-26 15:06:24,478 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-11-26 15:06:24,478 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-11-26 15:06:24,494 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLWriter
2020-11-26 15:06:24,494 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-11-26 15:06:24,516 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-11-26 15:06:24,547 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2020-11-26 15:06:24,563 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2020-11-26 15:06:24,578 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2020-11-26 15:06:24,594 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2020-11-26 15:06:24,616 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
2020-11-26 15:06:24,632 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeWriter
2020-11-26 15:06:24,663 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImageReader
2020-11-26 15:06:24,679 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
[... 117 lines cut ...]
2020-11-26 15:06:32,583 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2020-11-26 15:06:32,585 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1.0 NVIDIA 457.09
2020-11-26 15:06:32,589 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   NVIDIA Corporation
2020-11-26 15:06:32,592 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: GeForce GTX 1070/PCIe/SSE2
2020-11-26 15:06:32,595 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.0.0
2020-11-26 15:06:32,597 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\default.shader]...
2020-11-26 15:06:32,603 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\selection.shader]...
2020-11-26 15:06:32,609 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\select_face.shader]...
2020-11-26 15:06:32,614 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\default.shader]...
2020-11-26 15:06:32,619 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\composite.shader]...
2020-11-26 15:06:32,624 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\overhang.shader]...
2020-11-26 15:06:32,630 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\striped.shader]...
2020-11-26 15:06:32,636 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\transparent_object.shader]...
2020-11-26 15:06:32,640 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\striped.shader]...
2020-11-26 15:06:32,645 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray.shader]...
2020-11-26 15:06:32,650 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray_composite.shader]...
2020-11-26 15:06:32,655 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\default.shader]...
[... cut ...]
2020-11-26 15:06:32,676 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray.shader]...
2020-11-26 15:06:33,037 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 4.3s
2020-11-26 15:06:42,851 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

**Additional information**
a few prior versions of cura had been installed but the folders of each older version has been deleted in AppData\Roaming\cura and in ProgramFiles

nallath, 2020-11-26:
According to the log it took 7 seconds for Cura to start and 4.3 seconds for some other initialization. That seems to be pretty much on par with what I would expect.

Beerlord666, 2020-11-26:
> 
> 
> According to the log it took 7 seconds for Cura to start and 4.3 seconds for some other initialization. That seems to be pretty much on par with what I would expect.

i did some double checking (deleted all from stderr and let my stopwatch running as reference ): 
it took 31 seconds for cura from clicking on Cura.exe to the time when the gui/cura window got displayed.
i started the exe at about 16:09:00
the log starts at 16:09:23 ... What happened 22 seconds before the first log entry?

But to be honest, my real problem is that cura is freezing randomly and i cannot work with it.

so i opened a random .stl file at 16:17:00
cura froze. 
at 16:17:23 there was the next log entry.

so there is no log at all about the freeze

2020-11-26 16:09:23,190 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ConsoleLogger
2020-11-26 16:09:23,206 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraDrive
2020-11-26 16:09:23,221 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [92]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.8.0\CuraEngine.exe
2020-11-26 16:09:23,237 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraEngineBackend
2020-11-26 16:09:23,244 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileReader
2020-11-26 16:09:23,244 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2020-11-26 16:09:23,244 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: C:\Program Files\Ultimaker Cura 4.8.0\CuraEngine.exe
2020-11-26 16:09:23,259 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.8.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2020-11-26 16:09:23,259 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileWriter
2020-11-26 16:09:23,275 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJReader
2020-11-26 16:09:23,275 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJWriter
2020-11-26 16:09:23,290 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-26 16:09:23,290 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.8.0
2020-11-26 16:09:23,290 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2020-11-26 16:09:23,290 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-26 16:09:23,290 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2020-11-26 16:09:23,306 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-11-26 16:09:23,306 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-11-26 16:09:23,306 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2020-11-26 16:09:23,306 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-26 16:09:23,306 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2020-11-26 16:09:23,322 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2020-11-26 16:09:23,322 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLReader
2020-11-26 16:09:23,322 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2020-11-26 16:09:23,322 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2020-11-26 16:09:23,322 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2020-11-26 16:09:23,322 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2020-11-26 16:09:23,322 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-11-26 16:09:23,337 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-11-26 16:09:23,337 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-11-26 16:09:23,344 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLWriter
2020-11-26 16:09:23,344 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-11-26 16:09:23,359 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-11-26 16:09:23,375 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2020-11-26 16:09:23,391 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2020-11-26 16:09:23,391 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2020-11-26 16:09:23,406 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
[... cut ...]
2020-11-26 16:17:23,480 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\transparent_object.shader]...
2020-11-26 16:17:23,802 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectItemButton.qml:81:9: QML Button: Binding loop detected for property "width"
2020-11-26 16:17:33,274 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

nallath, 2020-11-26:
Before the first logs are shown, it tries to find the location of the plugins to load and some other similar tasks.

Do you happen to have a network drive available? It's a known issue that can lead to very long load times.

Alternatively; Have you tried disabling the Ultimaker network plugin (marketplace -> installed). On some systems this has also been reported to cause slowdowns.

Beerlord666, 2020-11-26:
> Before the first logs are shown, it tries to find the location of the plugins to load and some other similar tasks.
>
> Do you happen to have a network drive available? It's a known issue that can lead [...]

i disconnected all network drives, although no cura files are on them and it **worked**. 
cura is not freezing anymore. 
(only on first load of marketplace plugins after cura startup when i scroll down: 
2020-11-26 16:41:14,762 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles)

It seems like my issue is resolved.

Thank you very much!

closed 2020-11-26 by Beerlord666

Ghostkeeper, 2020-11-30:
See also https://github.com/Ultimaker/Cura/issues/1602 . The issue is only with the X: drive.