Ultimaker/Cura #8837: [4.7.1] On two color print with extruders 2 and 3, prime tower purges with extruder 1 before purging with correct extruder

lam4014, 2020-11-26:
Application version
4.7.1

Platform
Windows 10 64-bit, Intel i5, Intel UHD 620

Printer
Ender3 pro (4 extruders, 1 nozzle). All nozzle offsets are 0.

Reproduction steps

Printing 2 color STL model by opening 2 different partial modelsAssign extruder #2 to one STL modelAssign extruder #3 to other STL modelMergeEnable prime tower, 35mm, 80mm3, same values for all 3 extrudersSet Build Plate Adhesion Extruder to #2Slice

In my picture, I have extruder #1 loaded with black PLA, #2 loaded with red PLA and #3 loaded with white PLA.

Screenshot(s)
(Image showing the problem, perhaps before/after images.)

Actual results
Every time there is an extruder change, the current material is purged out by extruder #1. Only after doing so does it purge with the correct nozzle. You can see in the picture that the prime tower has black in it. This is from the #1 extruder, yet the #1 extruder is not assigned to any part of this print. There's basically two purges happening (ie #2 -> #1 -> #3 instead of just #2 -> #3) on every extruder change when only 1 should occur.

Expected results
Extruder #1 is not used in this print. Therefore, it should never be part of the prime tower. It should never be purged. It is simply wasting material unnecessarily as I need to increase the tower size/volume to accomodate the extra purge.

I have also tried disabling extruder #1 to see if I could work around this but Cura then says it can't slice because one of the assigned extruders is disabled.

I have searched all config options for any other options that include the word "Extruder" and there are none except for the adhesion extruder, which I already set to #2 (as can also be seen in the picture).

Project file

Na

Log file
N/A

Additional information
(Extra information relevant to the issue.)

Ghostkeeper, 2020-11-30:
I think I know why this happens. The prime tower selects one extruder as the extruder to use for the outer shell, which gets used as the shell for the entire print. This is to ensure that the prime tower has sufficient strength; different materials don't stick as well to each other (even different pigments sometimes).
The extruder selected is the material with the greatest Adhesion Tendency setting value (a setting that's not shown in the interface). In case of ties, which is probably the case for you since you use only PLA, it selects the material with the lowest extruder number. Which is extruder 1. However this "lowest extruder number" doesn't filter on which extruders are being used since that information is unknown at the stage of CuraEngine when this number is determined.

I'll ask my colleagues to see what they think about the priority of this bug.

Ghostkeeper, 2020-12-01:
It was not considered to be very important, considering the low (although increasing) prevalence of printers with >2 extruders on the market now (and Ultimaker not having any), and considering that the effect is just unnecessary extruder switches which increase printing time but don't do any real harm otherwise. We'll keep the issue open though in case priorities change (probably if Ultimaker would release a printer with >2 extruders) or in case an external contributor wants to work on it.

If someone wants to work on it: The `PrimeTower` class selects the extruder for the outer shell in its constructor. This constructor is executed by being a member of `SliceDataStorage`, in its constructor. At that time of constructing it is not yet known which extruders are being used. So it'd mean that logic needs to be moved and the extruder for the outer shell needs to be computed later.

There are two workarounds, both annoying in different ways:
* Use the first extruder in the print, i.e. switch filaments around in the printer.
* Give one of the used filaments a greater value for the Adhesion Tendency setting. However this setting is not shown in the interface so you'd need to edit it in the profiles with a text editor, which is error-prone.

Ghostkeeper, 2021-09-10:
This was reported again separately and nobody noticed it was a duplicate. This time when it got evaluated people did think it important enough though.

I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/9917. Let's track progress there.

closed 2021-09-10 by Ghostkeeper