Ultimaker/Cura #8838: [4.7.1] Glitched lines visible in gcode preview

knoellle, 2020-11-26:
**Application version**
4.7.1

**Platform**
Arch Linux (5.9.10-arch1-1)

**Printer**
Creality Ender 3

**Reproduction steps**
1. Loaded Model L-A and L-B from [here](https://www.thingiverse.com/thing:3033662/files) simulaneously
2. Slice (I'm using the Default Ender 3 on Standard quality. Only changed print sequence to `One at a Time`)
3. Go to preview and scroll through the layers.

**Screenshot(s)**
[image]
Object placement doesn't matter:
[image]
Nor does orientation
[image]

**Actual results**
Somewhere around the last ten or so layers, there are a bunch of lines going to the 0,0,0 point. It's not always on the same layer, but I have restarted cura a few times now and was able to reproduce this every time, although the line pattern is always different.
The lines only appear from one of the objects to the coordinate origin, but it's not necessarily the the object currently being printed in the viewed layer.

**Expected results**
There shouldn't be weird lines

**Project file**
Zipped, as github doesn't like .3mf
[project.zip](https://github.com/Ultimaker/Cura/files/5604962/project.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5604949/cura.log)

**Additional information**
This doesn't seem to affect the generated gcode file, loading it in gcode viewers doesn't show the lines.

Kveri, 2020-11-28:
I have the same issue. I'm printing Hero Me Gen 5 parts and using the following settings:
Low Quality (0.28mm layer height) default Cura profile with these overrides: wall line count = 3, bed temp = 115, combing mode = infill, support = yes, support placement = buildplate, adhesion = brim, print sequence = one at a time

First there is this at layer 170:
[image]

But when I go to layer 171 it changes to this:
[image]

And then all subsequent layers display properly. But when I go back from layer 171 to layer 170 this happens:
[image]

Then going back to layer 169 everything is normal again. I restarted Cura, reload file, remove all objects and re-inserted them. I can replicate it consistently.

When I disabled combing it changed to this (still errors):
[image]

Here are: cura profile, all 4 STLs, 3mf and also gcode.
[cura-issue-8838.zip](https://github.com/Ultimaker/Cura/files/5610651/cura-issue-8838.zip)

Ghostkeeper, 2020-11-30:
Indeed this reproduces. This seems to have something to do with it creating way too many layers for that second object which are all empty.

closed 2024-12-08 by github-actions[bot]