Ultimaker/Cura #8839: Patterns appearing on curves / corners - 4.8.0 and 4.7.x - not on 4.6.1

msteele999, 2020-11-26:
**Application version**
4.8.0

**Platform**
Windows 7 x64

**Printer**
Sovol SV01 (included with Cura 4.8.0 / copied from Ender 3 in 4.6.1)

**Reproduction steps**
1. Using 'identical' filament profiles and slicer settings, slice in Cura 4.6.1 and 4.8.0.
2. Print
3. Observe the difference between the Cura 4.6.1 sliced print (clean) and the 4.8.0 sliced print (patterns)

**Screenshot(s)**
4.8.0 results:

[image]

4.6.1 results:

[image]

**Project file**

[4.8.0_patterns_on_curves.zip](https://github.com/Ultimaker/Cura/files/5605261/4.8.0_patterns_on_curves.zip)

[4.6.1_no_patterns_on_curves.zip](https://github.com/Ultimaker/Cura/files/5605262/4.6.1_no_patterns_on_curves.zip)

**Additional information**

This was originally reported on the Glitch on Slow Downs issue - but that has since been closed as resolved - so this must be a different issue.

Mark

msteele999, 2020-11-27:
An additional picture of side by side

[image]

I stopped the 4.8.0 print due to the pattern showing up

bill-orange, 2020-11-27:
It could still be the same issue, I think.  For your particular machine you may have to change a couple of settings for 4.8.   In mesh fixes try setting Maximum resolution to 0.25mm and Maximum Deviation to 0.025 mm.  

Your layer height is pretty high.  Maybe drop that to 0.18 mm. If that doesn't help, I'm out of ideas.

msteele999, 2020-11-27:
I'll try the mesh changes, however I recall that causing other issues. 0.5
and .05 have fixed a lot of other problems for me.

The resolution test I'll try later today, however that really slows down
the print time.

Liger0, 2020-11-29:
Use a straight new profile, don't import old ones.

msteele999, 2020-11-29:
These are not imported. Made from scratch.

Liger0, 2020-11-29:
But is it a default 4.8 profile?

msteele999, 2020-11-29:
Yes. And that looked horrible. So then I started tweaking it to match my
4.6 profile, which looks phenomenal.

Ghostkeeper, 2020-11-30:
We can't really help you with print quality issues. Each print quality issue can have hundreds of causes. We've made thousands of changes between 4.6 and 4.8. We're unable to reproduce those print quality issues on our own printers with our own filaments and so we're unable to verify whether we fixed them if we make any new changes.

From your results I can't tell what Cura is really doing wrong here. Is there something unexpected in the g-code output of Cura? Are there any settings that you think should be changed in the defaults for your printer+nozzle+material+quality?

msteele999, 2020-11-30:
Thank you for your reply. It's very disappointing but I appreciate that you
did not give me the 'it's not an Ultimaker printer so go figure it out
yourself' response.

I have gone through literally 50 iterations of the profile, starting with
the default and making modifications to everything I could think of that
would affect the external skin of a print in a repeatable and obvious
pattern.

*When you printed the file using the 4.8.0 profile I supplied on your
printer, what were the results? Did you also see the pattern in the
perimeter that I showed in my pictures?*

I provided both the gcode and the project files for the model in 4.6.1 and
4.8.0 hoping you would see what is happening in the gcode - it's localized
to the one corner and I was hoping you could isolate the issue. I can get
the same results with multiple materials at multiple material settings. The
ONLY thing that resolves the issue is going back to 4.6.1

I understand if you are not willing to move any further with this issue as
it seems to be just one person having the issue. I will state that when I
use ideaMaker I don't see the pattern either - but I much prefer Cura and I
have invested a significant amount of time and energy tuning my printers,
profiles and materials in Cura. I will keep testing in between customer
print jobs.

Perhaps in 4.9.x it will revert back to whatever was working for 4.6.1

Best regards,

Mark Steele

Ghostkeeper, 2020-12-01:
> *When you printed the file using the 4.8.0 profile I supplied on your
printer, what were the results? Did you also see the pattern in the
perimeter that I showed in my pictures?*

I didn't print it. The settings used for the Sovol SV01 don't work on my printer. For one my printer has a Bowden tube and 2.85mm filament. And your layer height is not achievable in my printer. So anything I test will most likely end in either spaghetti or terrible print quality.

You had previously reported an issue with the wall overlap compensation sometimes being on one side and sometimes on the other. I don't really see how that could cause ringing, but moreover the compensation is very consistent when I try to reproduce that on my side:
[image]

So really, I don't know how I can help you. Other than by giving general printing advice, but my job is to develop Cura, not to help people with printing advice. Still, one thing you could look at is to change the infill pattern. Periodic banding like this is sometimes caused by having an infill pattern that is periodic in the Z direction. Instead, using an infill pattern that is consistent in the Z direction (like Triangle) could help.

msteele999, 2020-12-01:
Thank you,

I truly am not looking for printing advice - I fully understand that is not
your role. My hopes were that something would jump out and you would say
'Dude! turn off THAT setting there' and it would fix it.

I have tried changing the infill type, density and many settings. Even
taking it down to 5 % infill or putting it to 75% infile make no difference.

I'm looking now at the defaults that Cura uses when the printer is added,
vs. the ones that are added when I added the printer as an Ender CR-5 in
4.6.1 - maybe something in the settings there will lead me home.

For the time being, I run customer prints on 4.6.1 and experiment on 4.8.0
- If I find the fix, I'll report it back to you.

Best regards,

kiss81, 2020-12-04:
I share the same issues with 4.8. I can try what I want but I have the same issues (even worse actually). I tried imported, default profiles, lowered resolution but nothing helps. 4.6.2. works fine for me, so I kind of accepted to use that for now.

msteele999, 2020-12-04:
Question - are you using OcotPrint by any chance?

I have recently jettisoned OctoPrint and started printing straight from the
SD card.

In addition, I started with a stock 4.8.0 profile and made minor tweaks (#
of walls, infill type / density, combing and edge hiding) and have had much
better results. I will be testing the print in question later this
afternoon, but my preliminary results are starting with a stock profile and
making minor changes (but not touching resolution / width settings) in
combination with going to an SD card only solution, are very encouraging.

I'll post images later tonight.

kiss81, 2020-12-04:
No I print from the SD card. I print on a modified cr-10v2 (0.9deg stepper motors and hemera direct drive). 
I am using cura 4.8 on ubuntu by the way, I can give it a try on windows. 
The stock profiles from 4.6.2 print quite decent for me. I only need to change the usual "suspects" like temperatures and retraction distance / speed. I was hoping there was a nightly build available or something of 4.8.x / 4.9 so I can give that a try.

kiss81, 2020-12-06:
I have to get back on my previous comment: I am now printing with a default 0.20 preset and clean 4.8 installation on a windows  machine. All seems to be fine. So it's somehow an issue with the linux build for me or it's messing up my profiles somehow. I will now try to export my windows 4.8 profile and import that one into my linux build to see how that goes...

Ghostkeeper, 2020-12-08:
How did your print pan out with the Maximum Resolution and Maximum Deviation reset to defaults, @msteele999?

msteele999, 2020-12-08:
So - I have finally figured out what is happening.

Believe it or not, that is the z-seam.

I selected Sharpest Corner,  Smart hiding. Apparently in 4.8.0, it moves
the z-seam every dozen or so layers. I never considered this to be the
issue as in 4.6.1, the z-seam always stayed in the same spot.

If I have Cura 4.8.0 run thru the motion of the head on each layer via
preview, I see it move from the rear sharp corner (where it should be) to
the front of the model, where you see it in my pictures. After some layers,
it moves back. So I think this issue should be re-classified as 'Cura does
not keep z-seam placed at the same location when Sharpest Corner, Smart
Hiding is used.

The geometry of the model is the same on every layer, so the sharpest
corner is not moving - the z-seam should stay where it started.

Thoughts?

nallath, 2020-12-08:
It's one of those unintentional changes caused by something else. By changing the way that the simplify works, it changes the type of lines that end up in the model. Since the lines are different, it's also possible that it changes the decision where it places the z-seam (eg; same strategy, different input can result in a different outcome)

msteele999, 2020-12-08:
So would this be considered a bug or is this an opportunity for a new
feature: 'force z-seam to be in the same spot no matter what'?

Honestly, I'd love the ability to click on a spot on the model and say 'put
the z seam RIGHT there'

nallath, 2020-12-08:
That's already there: Set the "Z Seam Alignment" to "user specified"

msteele999, 2020-12-08:
Jaime,

OK - how about this then - put a visual queue in the preview of the seam
overlaying the entire model - don't make me go layer by layer to see where
it starts. Put a blinky marquee style line (just kidding) that shows where
the layer starts

Ghostkeeper, 2020-12-09:
As an aside, it's not just where the layer starts but where the contour starts. It should really just be called "seam" rather than "z seam".

We have a pull request that implements something like this: https://github.com/Ultimaker/Cura/pull/8599

msteele999, 2020-12-09:
Thank you for the link. This seems (no pun intended) like what I am looking
for. I'll follow that issue closely.

Best regards,

Mark.

closed 2024-12-08 by github-actions[bot]