Ultimaker/Cura #8840: Crua 4.8.0 keeps crashing on Mac OS 11.0.1 MacMini M1

supersoulman, 2020-11-27:
**Application version**
(The version of the application this issue occurs with.)

4.8.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

Mac OS 11.0.1 running on MacMini M1 Apple Silicon

**Printer**
(Which printer was selected in Cura?)

Prusa i3 MK3

**Reproduction steps**
1. (Something you did.)

after slicing I selected Preview and it crashed. 

2. (Something you did next.)

I have tried it 3 times with he same issue

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

it should not crash

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

not sure where to find it on Mac

**Additional information**
(Extra information relevant to the issue.)

supersoulman, 2020-11-27:
Process:               cura [22393]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.8 (4.8.0)
Code Type:             X86-64 (Translated)
Parent Process:        ??? [1]
Responsible:           cura [22393]
User ID:               501

Date/Time:             2020-11-27 13:46:09.918 -0500
OS Version:            macOS 11.0.1 (20B29)
Report Version:        12
Anonymous UUID:        3AE54068-6F22-6EE7-C0B5-B9A4AED93166

Sleep/Wake UUID:       5327F88A-247B-472F-8FC1-775FE632DFB1

Time Awake Since Boot: 58000 seconds
Time Since Wake:       400 seconds

System Integrity Protection: enabled

Crashed Thread:        24

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0:
0   <translation info unavailable>	0x00000001027a5f50 ???
1   GLEngine                      	0x00007fff6b9f2a12 gleUpdateDispatchJIT + 26
2   GLEngine                      	0x00007fff6b9f777a gleDoDrawDispatchCoreGL3 + 701
3   GLEngine                      	0x00007fff6b9c8909 glDrawElements_ACC_GL3Exec + 258
4   libQt5Quick.5.dylib           	0x0000000120a2b577 QSGBatchRenderer::Renderer::renderMergedBatch(QSGBatchRenderer::Batch const*) + 1111
5   libQt5Quick.5.dylib           	0x0000000120a2d345 QSGBatchRenderer::Renderer::renderBatches() + 1077
6   libQt5Quick.5.dylib           	0x0000000120a2e912 QSGBatchRenderer::Renderer::render() + 1922
7   libQt5Quick.5.dylib           	0x0000000120a1ea93 QSGRenderer::renderScene(QSGBindable const&) + 547
8   libQt5Quick.5.dylib           	0x0000000120a1e860 QSGRenderer::renderScene(unsigned int) + 48
9   libQt5Quick.5.dylib           	0x0000000120a5cfb4 QSGDefaultRenderContext::renderNextFrame(QSGRenderer*, unsigned int) + 68
10  libQt5Quick.5.dylib           	0x0000000120abd04f QQuickWindowPrivate::renderSceneGraph(QSize const&) + 783
11  libQt5Quick.5.dylib           	0x0000000120a4eabd 0x1209a4000 + 699069
12  libQt5Quick.5.dylib           	0x0000000120ac34fc QQuickWindow::event(QEvent*) + 460
13  PyQt5.QtQuick.so              	0x00000001207845aa sipQQuickWindow::event(QEvent*) + 106
14  libQt5Widgets.5.dylib         	0x000000010f8db0ba QApplicationPrivate::notify_helper(QObject*, QEvent*) + 314
15  libQt5Widgets.5.dylib         	0x000000010f8dd6ed QApplication::notify(QObject*, QEvent*) + 5181
16  PyQt5.QtWidgets.so            	0x000000010ed24d62 sipQApplication::notify(QObject*, QEvent*) + 114
17  libQt5Core.5.dylib            	0x000000010b8e3ae4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 164
18  libQt5Gui.5.dylib             	0x00000001105d4136 QWindowPrivate::deliverUpdateRequest() + 54
19  libqcocoa.dylib               	0x0000000121f1cc05 0x121efd000 + 130053
20  com.apple.AppKit              	0x00007fff232ad146 -[_NSOpenGLViewBackingLayer display] + 868
21  com.apple.QuartzCore          	0x00007fff26e725b8 CA::Layer::display_if_needed(CA::Transaction*) + 874
22  com.apple.QuartzCore          	0x00007fff26fac373 CA::Context::commit_transaction(CA::Transaction*, double, double*) + 517
23  com.apple.QuartzCore          	0x00007fff26e4ff91 CA::Transaction::commit() + 783
24  com.apple.AppKit              	0x00007fff22d539cb __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285
25  com.apple.AppKit              	0x00007fff2349dd0e ___NSRunLoopObserverCreateWithHandler_block_invoke + 41
26  com.apple.CoreFoundation      	0x00007fff204816c9 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
27  com.apple.CoreFoundation      	0x00007fff20481565 __CFRunLoopDoObservers + 549
28  com.apple.CoreFoundation      	0x00007fff20480a09 __CFRunLoopRun + 842
29  com.apple.CoreFoundation      	0x00007fff2047ffd2 CFRunLoopRunSpecific + 567
30  com.apple.HIToolbox           	0x00007fff28934fd0 RunCurrentEventLoopInMode + 292
31  com.apple.HIToolbox           	0x00007fff28934dcc ReceiveNextEventCommon + 709
32  com.apple.HIToolbox           	0x00007fff28934aef _BlockUntilNextEventMatchingListInModeWithFilter + 64
33  com.apple.AppKit              	0x00007fff22bcef85 _DPSNextEvent + 883
34  com.apple.AppKit              	0x00007fff22bcd74b -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1366
35  com.apple.AppKit              	0x00007fff22bbfbda -[NSApplication run] + 586
36  libqcocoa.dylib               	0x0000000121f2a001 0x121efd000 + 184321
37  libQt5Core.5.dylib            	0x000000010b8df2f1 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 417
38  libQt5Core.5.dylib            	0x000000010b8e41b6 QCoreApplication::exec() + 390
[... 1150 lines cut ...]
__UNICODE                          588K        1 
dylib (reserved)                  70.1G       18         reserved VM address space (unallocated)
mapped file                        4.1G     1065 
shared memory                       72K        6 
unshared pmap                     8096K        4 
===========                     =======  ======= 
TOTAL                             77.5G     5238 
TOTAL, minus reserved VM space     6.5G     5238 

Model: Macmini9,1, BootROM 6723.50.2, proc 8:4:4 processors, 8 GB, SMC 
Graphics: kHW_AppleM1Item, Apple M1, spdisplays_builtin
Memory Module: lp_ddr4
AirPort: spairport_wireless_card_type_airport_extreme, wl0: Sep 11 2020 18:25:33 version 18.20.222.10.7.8.97 FWID 01-17360706
Bluetooth: Version 8.0.1f5, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
USB Device: USB 3.1 Bus
USB Device: USB 3.1 Bus
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: USB Storage
USB Device: USB2.0 Hub
USB Device: Razer DeathAdder V2
USB Device: USB KEYBOARD
Thunderbolt Bus: Mac mini, Apple Inc., 0.0
Thunderbolt Bus: Mac mini, Apple Inc., 0.0

xereeto, 2020-11-30:
Also crashes when previewing on my MacBook Pro M1.

Ghostkeeper, 2020-11-30:
This is the relevant snippet from that Mac crash log:

```
Thread 24 Crashed:
0   ???                           	0x00007ffe95c829bc ???
1   libsystem_kernel.dylib        	0x00007fff2035c462 __pthread_kill + 10
2   libsystem_c.dylib             	0x00007fff2026cba3 raise + 26
3   libsystem_platform.dylib      	0x00007fff203ced7d _sigtramp + 29
4   ???                           	0x00007fff88b8bc48 __sCrossLinkPageSize + 8
5   libsystem_c.dylib             	0x00007fff202dd720 abort + 120
6   com.apple.AppleMetalGLRenderer	0x000000012552d97d invocation function for block in GLDContextRec::flushContextInternal() + 291
```

So it seems that your OpenGL drivers crashed.

I don't think it's really something Cura can do anything about. It should be noted that Qt also doesn't support your computer yet: https://bugreports.qt.io/browse/QTBUG-85279 (not even via Rosetta). As a result, we can't support it either.

I suggest that you try the compatibility mode for layer view, the option in the preferences screen. It restricts Cura to use OpenGL 2. Perhaps it doesn't hit the case that your OpenGL drivers crash on then.

ChristopheBis, 2020-12-08:
A big THANK YOU @Ghostkeeper !!
Disabling the option in the preferences solves the problem.

closed 2020-12-17 by Ghostkeeper