Ultimaker/Cura #8841: Install issue - Crashes right after Windows Firewall notice pop-up, Version 4.8

3dmakertech1, 2020-11-27:
Application version 
4.8.0

**Platform**
Windows 8 (x64) (build 9200)
2.40 gigahertz Intel Core i7-3630QM
12174 Megabytes Installed Memory
Intel(R) HD Graphics 4000 
c: (NTFS on drive 0) * | 197.57 GB | 49.82 GB free |  

**Printer**
Have not gotten that far yet...., 

**Reproduction steps**
Attempted to install, will not install past:
2020-11-27 12:23:23,964 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58
]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\xr
ay.shader]...

So, I took the following steps:
Uninstalled, re-installed, no change
Anti-Virus temp uninstall,
Made sure Graphics card updated {Intel HD Graphics 4000,  v9.17.10.2875)
Attempted previous versions 4.71, v4.7.0, v4.6.2...., no change, crashes at same point

**Screenshot(s)**
[image]
[image]

**Actual results**
Errors Reported:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	Cura.exe
  Application Version:	0.0.0.0
  Application Timestamp:	5f2c0247
  Fault Module Name:	ig7icd64.dll
  Fault Module Version:	9.17.10.2875
  Fault Module Timestamp:	507efe1e
  Exception Code:	c0000005
  Exception Offset:	0000000000736f28
  OS Version:	6.2.9200.2.0.0.768.101
  Locale ID:	1033
  Additional Information 1:	13c6
  Additional Information 2:	13c69dc4dfe090f2115ee509cf5082ba
  Additional Information 3:	6d3e
  Additional Information 4:	6d3e7cdf69c0e1be90d9d5f1210e9da6

***AND***

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	Cura.exe
  Application Version:	0.0.0.0
  Application Timestamp:	5f9933ba
  Fault Module Name:	ig7icd64.dll
  Fault Module Version:	9.17.10.2875
  Fault Module Timestamp:	507efe1e
  Exception Code:	c000041d
  Exception Offset:	0000000000736f28
  OS Version:	6.2.9200.2.0.0.768.101
  Locale ID:	1033
  Additional Information 1:	9753
  Additional Information 2:	975357ce700437e930aeec6397e81a1c
  Additional Information 3:	8597
  Additional Information 4:	8597539116c31af41a2bc30afb5d0479

**Expected results**
It should load, I want to really try this!

[... 280 lines cut ...]
erhang.shader]...
2020-11-27 12:23:23,879 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58
]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\st
riped.shader]...
2020-11-27 12:23:23,893 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58
]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\tr
ansparent_object.shader]...
2020-11-27 12:23:23,903 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58
]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\st
riped.shader]...
2020-11-27 12:23:23,918 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58
]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\xr
ay.shader]...
2020-11-27 12:23:23,927 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58
]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\xr
ay_composite.shader]...
2020-11-27 12:23:23,947 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58
]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\pl
atform.shader]...
2020-11-27 12:23:23,964 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58
]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\xr
ay.shader]...

**Additional information**
"Failed to load packages for Marketplace" is seen above...

doc-Seltsam, 2020-11-28:
https://github.com/Ultimaker/Cura/issues/8367#issuecomment-735233985

3dmakertech1, 2020-11-29:
Worked!!
Thank you!

closed 2020-11-29 by 3dmakertech1