Ultimaker/Cura #8843: Tree Supports not working as expected.

KenLucke, 2020-11-27:
**Application version**
4.62 through 4.8 (did not check earlier versions)

**Platform**
Mac OSX 15.5.7

**Printer**
N/A

**Reproduction steps**

1. Load a file with portion that needs support hanging in midair.
2. Slice using Standard supports - part is supported.
3. Reslice using Tree supports - part is not supported (I used several profiles to experiment, and tried changing virtually every setting pertaining to supports [including angles from 1° to 89°], with no change in result)

**Screenshot(s)**
Below  is a cross section of the model,  showing the "hanging" part, and resulting slices with Regular and Tree supports, showing the center is not supported when Tree supports  are enabled.

[image]

**Actual results**
Center "midair" part has no support

**Expected results**
part should be supported

**Project file**

https://www.dropbox.com/s/a6a9jmt5raw5j6j/Niger%20Thistle%20Sock%20Feeder%20%28Single%29%20Base.3mf

**Log file**

**Additional information**
(Extra information relevant to the issue.)

Asterchades, 2020-11-28:
Judging by the colours in your screenshots, you're using a Support Interface. There's known interaction issues between Interfaces and Tree Supports at present, meaning you'll likely see better results without the Interface.

KenLucke, 2020-11-28:
> Judging by the colours in your screenshots, you're using a Support Interface. There's known interaction issues between Interfaces and Tree Supports at present, meaning you'll likely see better results without the Interface.

I believe I mentioned that I changed virtually every setting in an attempt to solve this.  That includes using/not using a support interface.

Ghostkeeper, 2020-12-01:
The project file you provided seems to be a different model? It has no such small overhanging area.
[image]

Flipping the model upside-down sort of produces a little ring like you describe, but tree support seems to support that ring fine:
[image]

Could you share a project file that reproduces the problem?

KenLucke, 2020-12-01:
Oop, sorry.  I have the base and the cover for this niger seed bird feeder.  I shared the base accidentally.  Here's the cover:

https://www.dropbox.com/s/wro8nf5xuu3jpyu/Niger%20Thistle%20Sock%20Feeder%20%28Single%29%20Cover.3mf

KenLucke, 2020-12-01:
Here's another, similar one (this is for a larger, triple sock version of the feeder).  Does the same thing.

https://www.dropbox.com/s/e3638ejaeq40ml0/Niger%20Thistle%20Sock%20Feeder%20Cover.3mf

Ghostkeeper, 2020-12-03:
Yeah, with that project file the problem reproduces (I tried the first of the two anyway). I don't know what the difference is why this problem occurs with that model but not with the upside-down version of the previous model. But it does. I tried to change the settings that could influence this (just like you) but it never seems to want to support that little ring for some reason.

KenLucke, 2020-12-03:
> Yeah, with that project file the problem reproduces (I tried the first of the two anyway). I don't know what the difference is why this problem occurs with that model but not with the upside-down [...]

Glad it wasn't just me :).

Ghostkeeper, 2020-12-07:
(Developers, refer to CURA-7913.)

MariMakes, 2023-05-16:
Hey all,

Quick update from our side. 
To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we are cleaning up our open issues. I stumbled upon your issue with tree supports and I can confirm that will this will be resolved in the next release. 

We are introducing new tree support, and your floating island seems to be fixed. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-05-16 by MariMakes