Ultimaker/Cura #8844: [4.8.0] Make Overhang Printable closes up recessed regions in the bottom of the model

OpusMcN, 2020-11-27:
**Application version**
4.8.0

**Platform**
Window 10 x64, Geforce 1060

**Printer**
Custom FDM

**Reproduction steps**
1. Take model with a recessed area in the base where the recess could be printed without support (sample file attached)
2. Turn on Make Overhang Printable
3. Set the Maximum Model Angle to something larger than the angle of the recessed region (eg, 75 degrees for the sample file where the sides are approx 55 degrees)
4. Slice with support off (other settings can be set to sensible defaults)

**Screenshot(s)**
Result of slice showing bug
[image]
Expected result of slice
[image]

**Actual results**
The recessed hole in the base of the model is filled in when slicing

**Expected results**
The hole should be retained as the sides are not at a shallow enough angle to the changed by the "Make Overhang Printable" setting

**Project file**
[CFFFP_make_overhang_bug2.zip](https://github.com/Ultimaker/Cura/files/5609808/CFFFP_make_overhang_bug2.zip)

**Log file**
Included in attached zip file

**Additional information**
The included model has an additional recess removed from the side with exactly the same angles as the recessed hole in the bottom.  This side recess isn't altered but the Make Overhangs Printable setting which is the expected result.

OpusMcN, 2020-11-28:
A related issue (which may need to be a new bug report?) is that recesses with flat tops that can easily be bridged are also removed by Make Overhangs Printable.   I guess the top being flat means they pass the angle test, but it looks like some differentiation isn't being made between overhangs and bridges (which can be printed without modification)

Ghostkeeper, 2020-12-01:
Hmm, yeah I can see why that happens. It's a bit inherent in the way that this setting works, really.

Make Overhang Printable goes through the layers from top to bottom. Every layer is added to the layer beneath it with a small inset (with the inset distance depending on the angle). That's what creates the basic conical shape:
[image]

Because the layer above the tip of this recess is completely solid, it's going to cover up the hole completely, making that layer also completely solid. And this propagates all the way down.

Fixing this needs a bit of a re-think as to how the setting works.

This setting is currently expected to change the model everywhere. It doesn't detect bridging areas as that would cause a circular dependency. Changing the model to remove overhangs changes where the bridges are, and where the bridges are would then change where it removes overhangs.

OpusMcN, 2020-12-02:
Understood, thanks for the reply.  I wasn't sure exactly how it was calculated but that makes sense.  I can almost see a way to a solution where any closed interior contour on a lower layer gets removed from the correction.  You could push the contour of the hole back up to the higher layer (boolean subtract  them out) after the offset has been applied and then union the higher layer onto the lower one.   For nested contours on the lower layer you'd subtract/union them from the higher layer depending on their winding.   I haven't thought this through every detail but it looks like it'll get most cases (or at least the ones that would make my life easy :)   There might be cases where an enclosed hole might not get the correct "make printable" slope applied it all cases but it feels like getting it a bit wrong would be better than completely removing it.

closed 2024-12-08 by github-actions[bot]