Ultimaker/Cura #8845: Cura seem don't memorize nose sizes

GooodFr, 2020-11-28:
Hello,

Thank you so much for this free software !

I changed the nose of my printer (old is 0.4mm, new is 0.2mm)
So, in Cura, I created a new printer (based on Wanhao Duplicator i3 Plus)
In parameter of this new machine, extruder 1, I change the size of my new nose "0.2"
I close.
But if I slice, Cura don't care about my new nose 0.2 (2 lines = 0.8mm)
I back in parameter of the machine, extruder 1, and size back to "0.4"

Or I don't understand how to manage different size of nose …

I use Cura 4.7 under Windows 10 64b

Regards,

xjakubo, 2020-11-28:
hi, i also have this issue with my FlashForge Adventurer 3, but I found a solution; in nozzle size input type ',' instead '.', so for example '0,2mm' instead of '0,4mm'

GooodFr, 2020-11-28:
Great ! that work :)
Thank you 👍

Ghostkeeper, 2020-12-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/6546.

closed 2020-12-01 by Ghostkeeper