Ultimaker/Cura #8846: Gyroid infill sudden direction changes

Liger0, 2020-11-28:
**Application version**
4.8.0

**Platform**
Windows 10 x64

**Printer**
CR10S

**Reproduction steps**
1. Gyroid infill

**Screenshot(s)**
[image]
[image]

**Actual results**
Gyroid infill suddenly changes direction by 90° from a layer to the other, so the layers split as the bond is very weak, and the infill bridges

**Expected results**
Smoother gyroid transition between layers

**Project file**
[CCR10S_gyroid direction.zip](https://github.com/Ultimaker/Cura/files/5609876/CCR10S_gyroid.direction.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-12-01:
Honestly it looks accurate to me. You have a fairly high infill density of 30%, so the pattern is small. And a pretty big layer thickness of 0.28mm. At this infill density, the pattern oscillates in a space of about 3mm, which is 10-11 layers at your layer height. It's going to have to make quite big steps at every layer.

And the steps aren't very big in reality. It's bridging gaps of about 0.5mm.

Reducing the layer height shows a much smaller switching point:
[image]

Ghostkeeper, 2020-12-01:
I think the Gyroid pattern may just not be the right choice at great infill densities. You're not going to get a whole lot of flexibility out of the pattern then.

closed 2020-12-01 by Ghostkeeper