Ultimaker/Cura #8847: Retraction when Coming is enabled

viper0078, 2020-11-28:
**Is your feature request related to a problem? Please describe.**
When Coming is enabled, the Retraction process is disabled.
[image]
If the nozzle travels long distances without retraction, the filament will ooze out and the filament will be missing at the next printing start point.
In the picture below, the filament is missing at the start of the Innerwall and has failed to print.
[image]

**Describe the solution you'd like**
I want the option to perform Retraction when Coming is enabled.
Retraction prevents the filament from bleeding and prevents innerwall printing failures.
Coming improves print quality in many ways.

**Additional context**

This is what is described in the Combing Mode documentation.
-
If the start and end location are in completely separate paths, the nozzle will first move to the position where the two parts are closest together, then optionally retract (if retractions are enabled), travel to the target path, optionally unretract, and then move through the new part to its final destination. 
-
To prevent the filament from oozing, I think it's better to do a retraction before the nozzle starts to move.

Asterchades, 2020-11-28:
There's already user-configurable options for this.

"Max Comb Distance With No Retract" can be set to determine how far a combing move can be without also performing a retraction. If set to 0mm (the default) this will effectively disable retractions during all combing movements. But you can set this to any value you please, and it can be beneficial to set this to 1mm or less for prints with finer details (though I have heard people setting arbitrary values such as 10-15mm also works situationally).

There is also a "Max Comb Extra Distance" option which can be used to bypass combing in the event a movement is simply too long, which will fall back to a direct retract-move-prime sequence. This can also be used to avoid oozing from long combing moves such as traversing from one side of a print to another. As above, this defaults to 0mm which disables the functionality.

viper0078, 2020-11-28:
Oh really?
"Max Comb Distance With No Retract" was mistakenly thought to be the distance to disable Coming.
This is the ideal option for me.

I wonder where "Max Comb Extra Distance" is, or was it discontinued in V4.8?
Well, I don't use it even if it's there.

Thanks for the reply.

closed 2020-11-28 by viper0078

Asterchades, 2020-11-28:
> I wonder where "Max Comb Extra Distance" is, or was it discontinued in V4.8?

That one's on me. It's one of the additions Smartavionics made to his builds, and isn't in the base Cura.

Ghostkeeper, 2020-12-01:
> To prevent the filament from oozing, I think it's better to do a retraction before the nozzle starts to move.

It actually does. Only in your print that is all not applicable because it's staying within a single part and doesn't need to cross any walls.