Ultimaker/Cura #8849: Spiralize "Travel" Move on every layer

GregValiant, 2020-11-28:
**Application version**
4.6.2 and 4.8

**Platform**
Win10 Home 64 bit (running in Compatibility Mode)

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
1. Open a vase model in Cura that has no corners.
2. Slice in "Spiralize Outer Contour" mode.
3. View in Preview with "Travels" turned on.

**Screenshot(s)**
[image]
[image]
[image]

Cura Forum discussion:  [Link 30934](https://community.ultimaker.com/topic/30934-vase-mode-without-seam/?tab=comments#comment-264199) 
Cura Forum discussion: [Link 34966](https://community.ultimaker.com/topic/34966-will-cura-get-real-vase-mode-back/)

**Actual results**
The first screen shot shows the model used.  It is oriented the way the original poster had it.
The second screen shot shows the travel moves.  They occur at the end of every layer and the print head does a "back-up" before continuing on it's way.  This results in a noticeable seam in the print.

**Expected results**
Smooth movement as the print head ends and starts a layer.

**Project file**
[XYSeamInSpiralize.zip](https://github.com/Ultimaker/Cura/files/5610616/XYSeamInSpiralize.zip)

**Log file**
In the ZIP file

**Additional information**
This issue has come up twice now in the UM Cura forum.  The first time I was unable to repeat the results of the original poster.  This time there was no problem repeating the results.  The Travel moves cause the print head to shuffle back and forth as it finishes one "layer" and starts the next.  There is no Z component to the moves, they are in the XY only but they do produce a seam-like blemish in the print.

Let me know if you require further info,
Greg

smartavionics, 2020-11-28:
Try https://github.com/smartavionics/Cura/releases/tag/20201121, it will slice your project with no extraneous travels.

msageryd, 2020-11-29:
I just tried this version. The seam is still there. In fact, it’s even worse since the xy-jump seems to be several millimeters sometimes, which makes the seam zig-zag formed. 
[image]

smartavionics, 2020-11-29:
I have been printing one for 25 mins and can see no seam, will let the print complete and look closely at the result. Stay tuned...

smartavionics, 2020-11-29:
Well, the print was a failure but not because of a seam. I printed about 80% but it was suffering from bad layer shifting and finally it became a birds nest of filament. But I can assure you that the seam is not visible at all. Perhaps you have some setting that is causing the seam or maybe a script is having a detrimental effect.

msageryd, 2020-11-29:
That's great news. Could you please share your settings so I can give it another try?

I have found that this "vase" needs to be printed with the small diameter down. I.e. print the steepest overhang last. It seems that it's easier for the printer to cope with this overhang inwards, instead of outwards.

smartavionics, 2020-11-29:
My settings are nothing special at all, it would probably be easier if you could save your project file and then zip the .3mf file and attach to this issue so that I can take a look.

msageryd, 2020-11-29:
Here you go.

[vase.3mf.zip](https://github.com/Ultimaker/Cura/files/5612415/vase.3mf.zip)

- I'm printing this from nozzle 2, which is AA 0.8. 
- Material is Polymaker PolyMax PLA (similar to UM Tough PLA)
- I've printed this in both 0.1 and 0.2 layer height.
- Speed needs to be <25 mm/s in order for the steep overhang not to collapse.
- Leaving doors open on the printer for better cooling

smartavionics, 2020-11-29:
Well that was easy...

Surface mode = normal
[image]

Surface mode = surface
[image]

msageryd, 2020-11-29:
Woohoo!
Yes, it works! I have tried "normal" before. But I have tried soo many settings, I might just have tried "normal" in combination with some other settings, making it non-working.

Or.. is this possibly the fix in this version? I'll go ahead and try "normal" again in 4.8, to be sure.

But.. I'm curious. Why would "surface" leave a seam? The vase is a solid and I'm setting zero top and zero bottom layers. Wouldn't this effectively make "normal" the same as "surface" in this particular case?

That gcode viewer seems nifty. What's the name of it?

smartavionics, 2020-11-29:
> That gcode viewer seems nifty. What's the name of it?

It's the craftware slicer which lets you load up gcode files. Bit buggy but still very useful.

msageryd, 2020-11-30:
Confirmed. UM Cura 4.8 leaves a seam, even with surface mode = "normal".

Really great that you have fixed this. I do have a couple of outstanding questions.

- Why does "surface" still leave a seam?
- Will you make a PR for UM Cura out of this? Or do you thing UM will pick this up?

Ghostkeeper, 2020-12-01:
I'm not really sure what would be the expected behaviour when combining Surface Mode with Spiralize. Surface mode on "Surface" by nature doesn't combine with most other settings, since it removes any notion of volume from the model, and consequently also the inset operation. Instead it prints the surface triangles as single walls. Spiralize on the other hand creates a single inset wall, which requires a closed volume. The two don't really mingle. But apparently the Smooth Spiralized Contours setting still works fine on surface-mode lines so it seems to work okay.

The problem seems to occur in Cura only when Spiralize is enabled. Regardless of surface mode. The length of the overlap seems to be dependent on the Maximum Resolution setting so it can be assumed that the simplification algorithm has something to do with it.

GregValiant, 2020-12-02:
I had used Spiralize several times without issue and now there are these travel moves.  Going from Normal to Surface was my attempt to find something within the settings that would mitigate the seam.  Essentially I was randomly throwing switches to see which one worked the light.  It seems to have turned out that none of them did.

bitzl, 2021-12-05:
Are there any news on this?

GregValiant, 2021-12-06:
Using 4.12.1 I didn't have the problem with Spiralize turned on and Surface Mode set to normal.  I didn't try to print the vase but it looked right in the preview with no travel moves indicating that there would be a Z seam.

Ghostkeeper, 2021-12-16:
We didn't explicitly work to fix this issue. However in 4.9 there was a fix that could have been the same underlying issue: https://github.com/Ultimaker/CuraEngine/pull/1438

If you find that the issue doesn't reproduce any more we can close this issue. But I'm supposing Blitzl had a reason to post here, so maybe they still found a case where it's not going well.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

msageryd, 2022-02-07:
I (the one with the vase problem) can confirm that the seam is gone in Cura 13. It might have been solved earlier, but I didn't have time to test. I have printed my vase with both "surface" and "normal". It is still the case that you need to use surface mode=normal. 

I suppose UM picked up Smartavionics fix for this. Thank you @smartavionics. 

If you use "surface" there will be a seam. I still think it's odd that this leaves a seam, or at least it's odd that there is no warning if you use it together with "spiralize". I won't have a problem with this since I now know, but it will probably get people into trouble.

@GregValiant Thank you for your help creating the bug report. Please go ahead and close this issue.

GregValiant, 2022-02-07:
OK, I will.
Thanks @smartavionics and @Ghostkeeper and @fvrmr for the help and explanations.

closed 2022-02-07 by GregValiant