Ultimaker/Cura #8850: Print Setting Window

viper0078, 2020-11-28:
The print settings have so many items that it's hard to scroll through them.
I think it would be faster to divide it into tabbed pages for each section for quick access.

Depending on the settings, there are options that are hidden from view.
Instead of hiding them, why not gray them out?

Ghostkeeper, 2020-12-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/8121.

We have tabs where it's still possible to view multiple tabs at the same time. And we have visibility presets that limit the length of the settings list.

Only some people like to make all settings visible and open all tabs, and then complain that the settings list is too long.

closed 2020-12-01 by Ghostkeeper

viper0078, 2020-12-01:
The tab I'm imagining is shown below.
[image]

It depends on whether you need the visualization presets or not.
If people say it's too long, are they saying they don't need the parameters?
For those people, I guess you can use the presets.
I don't need the presets.
I can hide the ones I don't need by myself.

There is a search function, but for me it is not useful.
When adjusting parameters, I need to look at the values in each section from top to bottom to see if there are any relevant ones.
I don't use it in the list of search results because I'm likely to miss something.

I simply think it would be easier to access if each section was separated by tabbed pages, my personal preference.
I'm not complaining.