Ultimaker/Cura #8853: Support Blocker/Modiify Overlaps change not working with Fuzzy Skin

KenLucke, 2020-11-29:
**Application version**
4.8 (andd earlier)

**Platform**
Mac OSX 10.5.7

**Printer**
N/a

**Reproduction steps**

1. Load model
2. Select Fuzzy Skin
3. Use a support blocker/modify overlaps to turn OFF Fuzzy Skin in an area
4. Slice
5. Preview

[this has been confirmed both ways:  Main model Fuzzy Skin, Support Blocker/Modify  Overlaps Fuzzy off, and main model no Fuzzy, overlap with Fuzzy turned on..  In both cases the main model's settings are used for the entire model.

**Screenshot(s)**
[image]

**Actual results**
Entire model is still Fuzzy Skin

**Expected results**
Area within support blocker/modify overlaps should not bee fuzzy.

**Project file**
https://www.dropbox.com/s/7a5qexyssu7la6x/Fuzzy%20Bulb.3mf

**Log file**

**Additional information**

Ghostkeeper, 2020-12-01:
That's because your mesh is set to only modify infill. It'd create non-fuzzy walls inside the infill if the infill mesh wouldn't be set to have 0 walls (and that area of the mesh is too thin to have infill anyway).

Setting the mesh to be a cutting mesh causes it to affect the rest of the print as well instead of just infill.
[image]

This is expected behaviour, so we'll close the issue.

closed 2020-12-01 by Ghostkeeper

KenLucke, 2020-12-01:
OK, thanks for that explanation.  :)

mingyew1994, 2021-06-18:
Hi i'm having the same issue, i did change the Overlap to cutting mesh but the slicing isn't coming out with the fuzzy skin.
not sure if anyone is able to help me with this.

https://drive.google.com/file/d/1GbrfzBP5RIs3-oiFvDe2Rdu8kuL8Aw95/view?usp=sharing
https://drive.google.com/file/d/1b0KO0MhCBfMLffQmCKx9OWkkGlimPOAk/view?usp=sharing

Ghostkeeper, 2021-06-25:
Both those Google Drive links are not public. They require a log-in. I can't access them.

Fuzzy Skin still works fine as a per-object setting with a cutting mesh, from what I can tell.
[image]

mingyew1994, 2021-06-26:
Hi Ghostkeeper

could you try these 2 links? sorry for the trouble
https://drive.google.com/file/d/1GbrfzBP5RIs3-oiFvDe2Rdu8kuL8Aw95/view?usp=sharing
https://drive.google.com/file/d/1b0KO0MhCBfMLffQmCKx9OWkkGlimPOAk/view?usp=sharing

Ghostkeeper, 2021-07-05:
> Hi Ghostkeeper
> 
> could you try these 2 links? sorry for the trouble
> https://drive.google.com/file/d/1GbrfzBP5RIs3-oiFvDe2Rdu8kuL8Aw95/view?usp=sharing
> https://drive.google.com/file/d/1b0KO0MhCBfMLffQmCKx9OWkkGlimPOAk/view?usp=sharing

Your problem is that the Print Sequence is set to One at a Time. In that case, the cutting mesh doesn't work at all. The bug about that is here: https://github.com/Ultimaker/Cura/issues/9219

As a workaround, try setting that to All at Once.

mingyew1994, 2021-07-06:
Thank you very much ! sorry i didn't realise that was the issue !