Ultimaker/Cura #8854: Change Extruder Temperature per model for sequential prints

gunnir, 2020-11-29:
**Is your feature request related to a problem? Please describe.**
A number of filaments that I have (namely ColorFabb's LW series) require extrusion/flow _and_ temperature calibration, meaning a number of identical models must be printed at various temperatures. [ColourFabb recommend here](https://learn.colorfabb.com/print-lw-pla/) printing a series of test cubes at various temperatures and comparing the slicer line width to the actual extruded line width. They imply that taking the ratio of these numbers is the suitable flow modifier for that temperature. To do such, it would be convenient to add a nozzle temperature modifier per model setting, presumably only possible when 'print sequence' is set to 'one at a time' rather than 'all at once'.

**Describe the solution you'd like**
I've noticed between 4.6 and 4.8 this doesn't seem to be possible (despite this form suggesting it is) 
It appears this issue was actually tackled in the [following pull request](https://github.com/Ultimaker/Cura/pull/3154) for 3.3 onwards. However despite following the conversation advice in that pull, I still can't enable the per model setting of nozzle temperature for 4.8.

**Describe alternatives you've considered**
Thus far, I've attempted adding new machines, reinstalling Cura, following the advice outlined the the pull requested linked above. I cannot seem to initiate it from the UI - only if I edit the gcode post export can I enable this function.

**Affected users and/or printers**
Presumably anyone who attempted to print with ColorFabb's multivariate filaments that require temperature calibration. 

**Additional context**

Ghostkeeper, 2020-12-01:
The pull request you linked to was later reverted because of the hierarchy of how settings are requested in CuraEngine. In CuraEngine the settings override in the order of:

* Global
* MeshGroup
* Extruder
* Mesh

So adding temperature settings to the MeshGroup often didn't have any effect; they'd just get overridden by the extruder. The temperatures are obtained at extruder-level (see https://github.com/Ultimaker/Cura/issues/3193). Since it's impossible to have different sets of extruder stacks for different meshes, it is also impossible to print different meshes at different temperatures unless they are printing with different extruders. What you're asking for is indeed impossible in Cura at the moment.

Then the question remains whether we should make it possible.

Changing the nozzle temperature per-mesh is undesirable. This requires frequent adjustment of the target temperature of the nozzle, and PID controllers tend to freak out when you do that. By default, Cura limits temperature commands to once per 50s w.r.t. the stand-by temperature for this reason. The PID regulator is regularly 20 or 30 degrees off the target temperature if oscillating too quickly.

Changing the nozzle temperature per mesh group (=per mesh in one-at-a-time printing) is theoretically possible for the printer. Allowing this in CuraEngine though is a different business, due to the aforementioned architectural constraint. I don't think it's worth the effort right now. The main target audience of Cura is production, not calibration. You'd have to edit the g-code file for now to achieve this, or solve it with some complex search-and-replace scripts.

jeremysalwen, 2022-02-09:
@Ghostkeeper I am creating a plugin that uses optuna https://optuna.org/ to search for good Cura settings.  The idea is to create a grid of objects with a variety of print settings, and then rate the quality of the prints, and repeat, honing in on good settings.

I would very much like to be able to tune the extruder temperature per mesh group in this way, since it is one of the most important parameters to tune.  Is there any workaround?

Ghostkeeper, 2022-02-26:
A post-processing script could work around it. It can look for the `;MESH:xyz.stl` markers and change those. A simple option could be to add a search-and-replace script for `;MESH:xyz.stl\n` to be replaced by `;MESH:xyz.stl\nM109 S200\n` or something. It is not possible through CuraEngine alone.