Ultimaker/Cura #8857: Slice progress bar empty and stuck

DeviPirts, 2020-11-29:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
macOS Big Sur 11.0.1 - 16" Macbook Pro 2019 - Radeon 5500M

**Printer**
(Which printer was selected in Cura?)
Ender 3 PRO

**Reproduction steps**
1. Open the specific STL file
2. click the slice

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
[image]

**Actual results**
(What happens after the above steps have been followed.)
the slicing progress bar fills up, then empties and gets stuck. Cura never finishes slicing

**Expected results**
(What should happen after the above steps have been followed.)
Cura needs to finish slicing so that gcode can be saved 

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)
[DIPLAY_COVER_stockscrews.stl.zip](https://github.com/Ultimaker/Cura/files/5613129/DIPLAY_COVER_stockscrews.stl.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/5613131/cura.log.zip)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-12-01:
That sounds like CuraEngine crashes.

The file you provided is not a project file. It's just an STL file. And it slices fine on my side:
[image]

To create a project file, please click on File -> Save Project. This saves not only the models, but also the precise placement and all of the active settings and profiles. This way I should be able to reproduce your slice precisely, so hopefully it'll then also crash on my side so that we can see what the problem is.

golawojt, 2020-12-13:
I have the same issue.

**Platform**
Windows 10 64bit, Intel(R) Core(TM) i7-6600U CPU @ 2.60GHz   2.80 GHz, 16,0 GB RAM
Cura-mb-master-win64-20201121

**Project file**
[bug1.zip](https://github.com/Ultimaker/Cura/files/5684950/bug1.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5684964/cura.log)

nallath, 2020-12-14:
"2020-12-13 17:28:40,862 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [ERROR] Trying to retrieve setting with no value given: 'machine_always_write_active_tool'"

I can't really find what setting it's looking for though. It's not a setting that we currently support and we also didn't have that in 4.7

golawojt, 2020-12-14:
I removed files from `AppData\Local\cura\master` and now it works fine 
(I had earlier master from Cura-mb-master-win64-20200328 and I leave my settings during upgrade)

closed 2020-12-15 by no-response[bot]