Ultimaker/Cura #8859: Copy past of models

chrisje1980, 2020-11-30:
Hi Cura makers.

Is a Copy/paste option possible? 
That i can select a model and ctrl c ctrl v it :) 
That would be handy

Many thanks
greetings from the netherlands
Chris

fieldOfView, 2020-11-30:
No, you can not use Ctrl+C/Ctrl.V, but you can use Ctrl+M/Enter.

Ghostkeeper, 2020-12-01:
This used to be on our backlog (reported internally by myself) but it got removed from our planning as it was not deemed important enough. This was quite a while ago so maybe I'll propose it again then.

ragingsonar, 2022-01-10:
The multiply model option isn't a replacement for a simple copy paster imo; I never like autoplacement and will place things by hand. But it's most annoying when I'm trying to do an array of parts on "Print one at a time" and I just want to copy and offset a group of parts, but can't. 
Idk if I'm missing something.

Gravitate1, 2023-03-21:
Not sure if this was previously possible, but Cura 5.1.1 allows you to layout the parts as you wish. Then you can select all the parts, and group them (right click, "Group Models"). Then you can multiply the group, retaining the original layout.

MariMakes, 2023-10-11:
👋 Quick update on our side. 

This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-11 by MariMakes