Ultimaker/Cura #8860: GCode Layer marker not at new Z height

DanHudson15, 2020-11-30:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

DanHudson15, 2020-11-30:
I noticed that the generated G code does not have a new Z height at the LAYER marker in the code, its usually ~13 lines above with a ;MESH:NONMESH marker above it.

Ghostkeeper, 2020-12-02:
Yeah, the `;LAYER` comment is expected to be placed where it actually starts printing a layer. There are some preparatory travel moves that move the nozzle from the last position of the previous layer to the first position of the next layer. Those are currently considered part of the previous layer.

You'd like to have those be considered part of the next layer, I'm assuming? Is there a problem you'd solve with this change? Perhaps you could fill in the issue template for us?

closed 2020-12-16 by no-response[bot]