Ultimaker/Cura #8864: Profile Manager errors

andrewjm1959, 2020-12-01:
**Application version**4.8.0
(The version of the application this issue occurs with.)

**Platform**Windows 10
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**Ender 3
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**Version does not copy over all profiles in custom profiles from previous versions. Version does not let you import CURA.ini files which I have done with other versions. Last version known to do this correctly is V 4.0.0
[image]
[image]

(Extra information relevant to the issue.)
First photo is 4.0.0 with original profiles and second shot shows that only 2 have been carried over with 4.8.0

Ghostkeeper, 2020-12-03:
It's a display issue that is confusing you here. The second screenshot is showing a bug since it's showing settings for certain extruders in the global tab. The first screenshot is showing it correctly. Only global settings are visible in the global tab.

The display issue that is confusing you is this one: https://github.com/Ultimaker/Cura/issues/7626 . This is only a display issue. The settings should appear correctly in the main settings list.

andrewjm1959, 2020-12-03:
Sorry Ghostkeeper but, the missing PROFILES are also as a Cura.ini file and they are not reconised when i try to import them. I have attached Cura.ini profiles, try for yourself. These are from 3DLabPrint, they work fine on all versions 4.0.0 and earlier only.
RegardsAndrew

 
 

Closed #8864.

Ghostkeeper, 2020-12-07:
> I have attached Cura.ini profiles, try for yourself.

You didn't attach them. Or at least, maybe you attached them to the e-mail but e-mail attachments don't get posted on Github so I can't see them then. Could you go on the Github website and attach those profiles then?

Please fill out the issue template. It helps prevent... misunderstandings. We'll need reproduction steps, a log file and any files you might be importing for this one.

andrewjm1959, 2020-12-07:
[CURA_setting.zip](https://github.com/Ultimaker/Cura/files/5652154/CURA_setting.zip)
[CURA_setting.zip](https://github.com/Ultimaker/Cura/files/5652157/CURA_setting.zip)

closed 2020-12-07 by andrewjm1959

Ghostkeeper, 2020-12-08:
Those profiles seem to import fine for me.
[image]