Ultimaker/Cura #8866: Bridging: "bridge has multiple layers" option doing nothing

Fjordx, 2020-12-01:
**Application version**
(The version of the application this issue occurs with.)
4.8, 4.7, 4.5 ..?
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 Professional, radeon 5700xt
**Printer**
(Which printer was selected in Cura?)
ender 3
**Reproduction steps**
1. set : enable bridge settings
2. set: bridge has multiple layers

**Screenshot(s)**
[image]

**Actual results**
The first bridge layer is set with the appropriate parameters, the following are printed with default parameters

**Expected results**
The following two layers should be printed with less material and faster. This was seen in real as well as on online gcode analyze tools

**Project file**
https://we.tl/t-tM5zCeJTMR

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[cura.log](https://github.com/Ultimaker/Cura/files/5623930/cura.log)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-12-03:
This is not a bug but a feature gap. You adjusted the density, flow rate and speed **of the skin** in second and third layers. However there is no skin in the second or third layers. It's all walls.

The problem is that there is no setting to adjust the wall flow or speed in the second or third layer of bridging, and the setting that adjusts it for the main bridging layer doesn't adjust it for the extra bridging layers.

Fjordx, 2020-12-04:
But the walls are the outermost part which is always visible. would it not make sense to print them with a smaller density too to counteract sacking? This is the main problem for long overhangs imho.
just think about the saved filament for supports too, think about the environment! ;)

Fjordx, 2020-12-04:
But the walls are the outermost part which is always visible. would it not make sense to print them with a smaller density too to counteract sacking? 
How do I provide the devs with a change request that it has a good chance to get approved?
This is the main problem for long overhangs imho.
just think about the saved filament for supports too, think about the environment! ;)

Ghostkeeper, 2020-12-04:
> But the walls are the outermost part which is always visible. would it not make sense to print them with a smaller density too to counteract sacking?

Maybe. Walls are also printed with a lower speed though so perhaps the problem isn't as bad with walls? This feature was added through a pull request from a third party, so it didn't get any research from us on what should be included in the feature.

We've discussed the priority of this issue this morning, but decided that we probably can't get to work on this in the next half year or so, so to be realistic we have to defer this and not plan in any fix yet. The reason for this choice is mainly the usage statistics of these settings (which is a chicken-and-egg problem of course) and the fact that it seems to print fine without second/third layers for the most part as far as we've seen on our printers.
If we do a push to bring the bridging settings out of experimental, like we did for Tree Support in 4.7, then I can see this sort of issue getting attention too.

Fjordx, 2020-12-04:
Wow, thanks for that elaborate answer! 
I have 50mm long bridges, printing them with supports would take a huge ammount of time and filament bc of the height.
I"ll cross my fingers and gcode in the meantime for it to happen ;)

mkosch, 2022-12-22:
Bridging walls with greater wall line width than nozzle size doesn't work well for me, for example a 0.65mm wall line with with a 0.4mm nozzle size.

I can adjust the wall flow for the first bridge layer, which helps. However I cannot adjust the wall density to compensate for the much lower flow, so the bridged walls show too much space in between and don't thick together well.
For the second bridge layer the walls then get printed with the full flow of the bigger line width and sag below the first bridge layer and the skin doesn't stick well to the walls.

For my settings it would be very helpful to either adjust the wall density for the first bridge layer, which would then result in more wall lines or alternatively to adjust the wall distance, so that the bridge effectively would use a smaller wall line width and bigger skin area.
For the two layers above the bridge it would also be necessary to adjust the wall flow, wall density (or distance) and possibly also the wall speed, because one bridge layer is not enough to withstand the pressure of the greater wall line width.

**Here is another idea:**
What about printing bridges without traditional walls and skins? For the first bridge layer the skin pattern is not used at all, so this seems to me as a different type of line. The orientation of the lines depend on the bridge orientation and the whole bridge surface could be printed with the same line structure which would make it easier to adjust the density and flow for the whole surface. For the two layers above the bridge this approach would also be useful, but the line orientation should be changed (for example +/- 45 degrees relative to the bridge line orientation) to glue the bridge lines below together. This would be similar than printing skin without surrounding walls.

L-Pellegrini, 2023-03-18:
Came here to find out how it was possible that this feature is not yet implemented, as it seems not to be so complicated to do.
So... +1 to this feature request.
Thanks