Ultimaker/Cura #8867: Object Shadow counted as object size in Bed limit

TheCuraMan, 2020-12-01:
**Application version**
4.8.0

**Platform**
windows 8

**Printer**
Ender 3 Pro

**Reproduction steps**
rearranging object position on bed 
 
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
Before:
[image]
After:
[image]

**Actual results**
When shadow of the object gets out of the bed limit it is counted as part of the fiscal object so Cura wont slice even dough there is a lot of spear room to play with.

nallath, 2020-12-02:
Could you also share the model? It's either the calculation of the hull that had some issues, or the model has a very small floating bit that is included in the hull.

TheCuraMan, 2020-12-02:
> 
> 
> Could you also share the model? It's either the calculation of the hull that had some issues, or the model has a very small floating bit that is included in the hull.

Hello. It is to big to share it here. But you should be able to download it here https://drive.google.com/file/d/1CsV7kMKAb1-k_-zEqSIo8YUBZJNdTomg/view?usp=sharing Let me know if it works for you.

I understand what you mean but I positioned the object in different positions and the issue happens in all of them where the shadows on the bed floor is longer than the fiscal object.

I tried to replicate the issue with other objects but it was difficult to find other object that project a shadows that does not match the boundaries of the fiscal object.

For some reason with this object in particular the shadow projected can be longer than the fiscal object. Other object have the shadow right under them but in this case it looks like having the light source at a side instead of over it.

konskarm, 2020-12-03:
The shadow that you see is called the [convex hull](https://en.wikipedia.org/wiki/Convex_hull#:~:text=In%20geometry%2C%20the%20convex%20hull,convex%20set%20that%20contains%20it.) of the object. If any part of this convex hull (which encloses the entire object) is out of the buildplate or touches disallowed areas, it means that the printer head won't be able to print that part because it cannot reach to that point.

As @nallath suspected, the reason that there is such a big shadow (convex hull) is because your model contains some tiny little dots at that point, which technically are part of the object itself.
[image]

So if you move the object to the side, these dots will be outside of the buildplate and the printer won't be able to print them, hence why the entire object is marked as non-printable. You will need to edit your model in a 3D modeling software and delete these dots.

All-in-all this is a model error, so I am closing this.

closed 2020-12-03 by konskarm