Ultimaker/Cura #8868: [3.5.1] Slicing Engine Crash for Cura

thalesalacobra, 2020-12-01:
**Application version**
3.5.1

**Platform**
Windows 10 / GPU: Intel(R) HD Graphics / Dell Inspiron 3147

**Printer**
AnyCubic i3 Mega

**Reproduction steps**
1. Open Cura
2. Load .stl file
3. Begin slicing

**Screenshot(s)**
Slicing object 
[image]

Slicing progress emptied
[image]

**Actual results**
-Slicing progress bar reaches halfway
-Bar empties and does not restart

**Expected results**
Program finishes slicing and presents object to save in .gcode for printing.

**Project file**
https://drive.google.com/file/d/1CS_BOkgcEU0b5lkkUECdxi450OZNkH_M/view?usp=sharing

**Log file**
https://drive.google.com/file/d/18iJkyOTHs52kUzwaw9BYhRL2FlMbxRNh/view?usp=sharing

**Additional information**
I have confirmed that this is not a RAM issue. When slicing, and only with this program open, the slicing engine peaks at 900 MB, which kicks memory usage up in the 80% range. Other, larger, projects will tax the RAM harder, but still finish slicing after a period of time, which I have confirmed after having this issue with this particular project.

nallath, 2020-12-02:
3.5.1 is a pretty old version of Cura. Have you tried if this still an issue with the latest release (4.8.0)?

thalesalacobra, 2020-12-02:
There sure is egg on my face. I've updated and it works perfectly.

I'd tried updating at both the 4.3 and 4.6 versions, which both failed to launch on my PC for some reason, but I honestly didn't think to check for a more recent update. Sometimes, it is the most obvious things that fix the issue.

Thanks.

closed 2020-12-03 by konskarm