Ultimaker/Cura #8869: Minimum infill length

BixlerVDL, 2020-12-01:
Hello everyone, I hope your all healthy.

I have tried to print something with a 2.5mm thick outside. Now, because my wall thickness is set to 1.2mm (0.4mm nozzle so 3 walls total), The outside will consist of two 1.2mm thick walls and 0.1mm of infill space. (2.5-1.2-1.2=0.1mm) this space is so small that I don’t need the infill, in fact it takes so much longer now because of constant acceleration, deceleration.... I came across a feature called minimum infill area. This won’t work for me, I need infill in places which have large overhangs, 0.1mm doesn’t need infill. 

The solution hopefully would be a feature which would allow me to set the minimum distance of an infill line from wall to wall. This way, the tiny gaps between the walls don’t need infill whereas the large gaps do.

Sorry if it’s a bit confusing, I can show pictures if I need to clarify, please let me know.

Ghostkeeper, 2020-12-03:
There's a workaround for this: enable the Connect Infill Lines setting.

Ghostkeeper, 2020-12-03:
Perhaps a similar solution to https://github.com/Ultimaker/Cura/issues/8761 can be applied here too.

For some prints the lines are desirable since they connect the two opposing walls that could otherwise slide past each other.

BixlerVDL, 2020-12-04:
> There's a workaround for this: enable the Connect Infill Lines setting.

Unfortunately that had no positive effect, I'll try the #8761 solution later today. I'm trying to enable and disable infill in different parts of the same object, where the distance between walls is small I want no infill and when the distance is larger than x I want it to make infill.

BixlerVDL, 2020-12-06:
#8761 did not solve my problem.

Ghostkeeper, 2020-12-07:
That'll be because #8761 is not implemented yet.

BixlerVDL, 2020-12-07:
Correct. It’s a simplify3d feature, I don’t have that.

BixlerVDL, 2020-12-07:
Will the new feature be developed for Cura?

Ghostkeeper, 2020-12-08:
It's on our backlog somewhere. I give no guarantees, but there is intention.