Ultimaker/Cura #8870: Preferences and some UI widgets are unreadable when desktop uses dark theme

MarcinOrlowski, 2020-12-01:
I am using Cura 4.8.0 on Kubuntu 20.04 and my desktop is KDE with dark global theme "Breeze Dark", and some Cura UI elements are drawn unreadable due to colors used. The issue is not about Cura theme as the problem is the same no matter what Cura theme is set and it looks like the issue is about not having all the UI colors set, so theme's color are used and this is problematic due to theme being dark. 

Main window looks all fine and the problem is mostly visible in Preferences (but not all settings panes are affected). 

Main windows section list is affected:

[image]

All inputs are affected (see filter which should read "invisible", also "Currency" field on previous screenshot is also not empty)

[image]

Materials are hardly usable

[image]

Profiles render settings incorrectly

[image]

elektrikfisch, 2020-12-01:
Same issue here on openSUSE Tumbleweed.

Problem also existed in 4.7.1

elektrikfisch, 2020-12-01:
Changing the system's Global Theme from _Breeze Dark_ to _Breeze_ and restarting Cura results in a usable settings interface.

It seems as though some elements of the settings interface will change with the system theme, while others remain stuck on the default. The Material Information tab, for instance, remains stuck with a light grey background while the text also changes to a light grey color due to the system theme.

MarcinOrlowski, 2020-12-02:
I believe this issue is most likely about not setting all the colors used in UI and not about following system theme. Someone most likely forgot to set all the colors. This often happens when i.e. you set text input background color to white, but forgot setting text color, so your app widgets use default color, which comes from system theme (and is dark). As long as you use light theme, everything may look fine. Once you switch system to dark theme, default theme text color changes to white to contrast with theme's dark background color. But your app still sets only background color (to the same white) so widgets again inherit text color from system theme but this time it is light one.

Ghostkeeper, 2020-12-03:
Duplicate of https://github.com/Ultimaker/Cura/issues/5490.

closed 2020-12-03 by Ghostkeeper