Ultimaker/Cura #8873: camera zoom origin bug?

yzorgone, 2020-12-02:
I am on a new Macbookpro (2019) With Big Sur OS.

Cura release. 4.8

Ultimaker S5

Problem:
Cura operator camera zoom zooms to an irrelevant point in the left corner of the printer.
When Selecting the object the camera resets view to center. but the zoom point stays offset to that left corner. This results in: when zoomed in, the Camera rotates not around the desired object anymore.

Zoom problem video:
https://youtu.be/LcYjNyKO2lA

nallath, 2020-12-02:
You might have "Zoom towards mouse direction" enabled in preferences. Could you check if that's the case?

yzorgone, 2020-12-03:
> You might have "Zoom towards mouse direction" enabled in preferences. Could you check if that's the case?

Thanks!
that fixed it. its now back to the old behaviour.

closed 2020-12-04 by Vandrasc